QIDITECH/QIDIStudio #1: OMG - is QIDI AMS coming?

TinkerBarn, 2024-09-07:
Looks very nice! Especially the support for more than one plate! I asked for it in the past.

Perfect!

and it seems that the QIDI AMS is hopefully coming in near future, since you can set for each object on the plate a different filament. 

Please make me happy with it 🙏👍😄

Hukete, 2024-09-10:
> Looks very nice! Especially the support for more than one plate! I asked for it in the past.
> 
> Perfect!
> 
> and it seems that the QIDI AMS is hopefully coming in near future, since you can set for each object on the plate a different filament.
> 
> Please make me happy with it 🙏👍😄

Good work, looking forward to it.

designgears, 2024-09-27:
[image]

closed 2024-10-01 by TinkerBarn