QIDITECH/QIDIStudio #2: Printing over wifi does not work!!!

TrollPlayer94, 2024-09-09:
Connecting the device on QidiStudio with wifi does work and the fluid menu does appear in the device tab, but when you slice a file and try to send it over wifi it simply gives you a error. Will this be fixed in the next release or is this just happening to me (never had this problem on QidiSlicer).

QIDITECH, 2024-09-10:
Please tell me some more details. What went wrong, which model did you choose, and how did you operate? Please include a video or picture to clarify the problem.

RCsForVetsPREZ, 2024-09-11:
> Connecting the device on QidiStudio with wifi does work and the fluid menu does appear in the device tab, but when you slice a file and try to send it over wifi it simply gives you a error. Will [...]

Mine is doing the same

RCsForVetsPREZ, 2024-09-11:
> Mine is doing the same

Q1 PRO. QIDI SLICER  send via wifi to print and it will time out

QIDITECH, 2024-09-16:
Ok, we will fix this issue in the next version.

QIDITECH, 2024-09-30:
This bug has been fixed in the latest release.

closed 2024-09-30 by QIDITECH

RCsForVetsPREZ, 2024-09-30:
Ok thankyou I will try to update it again. I just did it yesterday and it
still disconnects… for now I’ve move to LAN