QIDITECH/QIDIStudio #6: Feature request.... Drag and Drop

ZAKsPop, 2024-09-23:
I would like to see drag and drop. I'm used to just dragging files onto the build plate from my desktop or file folder in Orca.

QIDITECH, 2024-09-23:
If you open the software directly from the installation package, this problem will occur due to insufficient permissions. After restarting the software, this problem will not occur again.
But this is slightly different from your description, I wonder if it is caused by this problem.

Burnstation3D, 2024-10-02:
Also will not allow opening files from network mapped drives / NAS from shares in my Pc in windows. Can access via network share tab but i didn't map out all my drives to have to do that everytime. drag and drop is also issue locally and from explorer tab from mapped drives. its how my workflow has been with orca for over a year so not sure why its been messed with in qidi studio.