QIDITECH/QIDIStudio #7: Crash with multiple displays

marekdovjak, 2024-09-26:
I have PC with 2 displays with different resolution and DPI

1. Open QIDI Studio
2. Drag the application window from one display to another 
3. The application freezes for a few seconds and then crashes

QIDITECH, 2024-09-27:
Sorry, our software is not compatible with multiple screens at this time.
This problem cannot be solved for a short time.

jaydenseric, 2024-10-16:
This is a very bad issue:

1. Create a new project.
2. Change some of the strength settings in Prepare -> Process -> Strength.
3. Attempt to close the window without saving. There will be a popup.
4. In the last popup that shows what settings were changed and will be lost, drag that popup onto your other monitor.

The popup just disappears, with no way to get to appear. But the main project window still seems to think it's open, and doesn't let you interact with the ability to close the window, etc. You are just stuck. I'm about to force quit the application to try to resolve this.

imperator-maximus, 2024-10-24:
this is quite a show-stopper. Any ETA for a fix here?

QIDITECH, 2024-11-16:
It has been confirmed that if the graphics card or resolution or PDI is different between the two screens, the software may not work properly on the second screen. But no solution has been found, please wait.

MCCVargues, 2025-02-15:
Hey, I have the same issue.

The program crashes and closes itself when I move the window from one screen to another. My main monitor is 4K and the second is 1080p.

closed 2026-11-05 by wjyLearn