QIDITECH/QIDIStudio #8: Freeze and crash in the device tab

bednarekpiotr, 2024-09-26:
I'm facing issue that the window is freezing and after few seconds it ctrashes completly. 
I have also error "Klippy not conected" 
I'm using [QIDIStudio_1.9.5.51_Linux.AppImage] on manjaro 

[image]

origin2000, 2024-10-05:
See the same issue with openSUSE Leap 15.6. No problems with ORCA 2.2.0.beta2.

breathless19, 2024-10-05:
My QidiStudio crashes instantly when trying to add a device at all via the local tab.   I click on the + button, input the ip config, press test, test confirms, click ok and the program locks up and crashes to desktop.   The program then cannot be reopened at all unless the json config file for the "newly added" printer is deleted.   Then the program can be reopened, but of course the printer is not added.   This is on Fedora 40 (bazzite), and the same exact problem occurs in other distros via distrobox containers.

origin2000, 2024-10-06:
Just tested with Mint22, which is an Ubuntu 2404 flavor. 
./QIDIStudio_1.9.7.52_Linux.AppImage
/tmp/.mount_QIDIStYU7dsG/bin/qidi-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

OrcaSlicer_Ubuntu2404-beta2 no problem.

QIDITECH, 2024-10-07:
QIDISlicer and QIDIStudio now depends on WebKit library, which greatly complicates its distribution. Latest Linux distributions (such as Ubuntu 24.04, Fedora 40) ship with newer version of WebKit than older (but still supported) distros. Bundling WebKit into the AppImage is difficult and may not be possible.
This has also led to a variety of different issues in different linux versions, which we will address in future updates. Please understand that the problem is too complicated to be solved quickly.

origin2000, 2024-10-07:
Thanks for the reply. I would like to provide some more input:

- OpenSUSE Leap 15.6 not only provides libwebkit2gtk 4.1 but also libwebkit2gtk 4.0. Therefore QidiStudio can be launched, but unfortunately core dumps in the device tab, when trying to connect to fluidd.
- Ubuntu 2404 (i.e. Mint22) by default only seem to provide libwebkit2gtk 4.1.which results in a core dump immediately at start cause the Appimage requires 4.0. I maged to install a 4.0 version from the Ubuntu Archive. So I can launch QidiStudio, use it for slicing, but as soon I use the device tab it even shows the fluidd interface but then core dumps again.

Given the fact that ORCA currently has working Linux versions and Qidi still needs some time I would highly appreciate if Qidi could provide the **full printer & filament presets** for import into ORCA

nattymcfly5, 2025-03-09:
@QIDITECH Is there any update on the next release that contains the fix for this? Looks like its this ticket and [issue 46](https://github.com/QIDITECH/QIDIStudio/issues/46)

breathless19, 2025-03-10:
> We are working to fix the problem. You can use the following command line to run QIDIStudio to keep the device TAB working properly.
> 
> ```
> WEBKIT_DISABLE_COMPOSITING_MODE=0 WEBKIT_DISABLE_DMABUF_RENDERER=0 ./QIDIStudio_1.10.02.76_Linux_Ubuntu_24.AppImage
> ```

Does this work on QidiSlicer as well?

wjyLearn, 2025-03-11:
> > We are working to fix the problem. You can use the following command line to run QIDIStudio to keep the device TAB working properly.
> > ```
> > WEBKIT_DISABLE_COMPOSITING_MODE=0 WEBKIT_DISABLE_DMABUF_RENDERER=0 ./QIDIStudio_1.10.02.76_Linux_Ubuntu_24.AppImage
> > ```
> 
> Does this work on QidiSlicer as well?

Available for QIDISlicer in Ubuntu 24.

nattymcfly5, 2025-03-25:
> We are working to fix the problem. You can use the following command line to run QIDIStudio to keep the device TAB working properly.
> 
> ```
> WEBKIT_DISABLE_COMPOSITING_MODE=0 WEBKIT_DISABLE_DMABUF_RENDERER=0 ./QIDIStudio_1.10.02.76_Linux_Ubuntu_24.AppImage
> ```

Thank you for replying and for the work on Linux.

This did not fix the problem for me. The only thing that worked was 

`env __GLX_VENDOR_LIBRARY_NAME=mesa  __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json `

but when I slice the render time is very very slow, I cannot maneuver the model.

Xephier102, 2025-04-21:
> QIDISlicer and QIDIStudio now depends on WebKit library, which greatly complicates its distribution. Latest Linux distributions (such as Ubuntu 24.04, Fedora 40) ship with newer version of WebKit [...]

Question; You must have made hundreds of thousands, if not millions selling these printers by now. So.. How much can it really cost to hire on a programmer to program you an app that works with this printer, instead of continuing to use an app with pieces copied from other printer's software? I can understand doing that as a startup, but as best I can tell, you guys have been around for years at this point. 

Yes, it would cost money, but it would easily be worth it when you've got a software that you're able to keep 100% functional 95-98% of the time. Cuz then people say more good things about Qidi, and maybe more importantly, less bad things about Qidi. 

Also, for the love of all that is holy, on the Qidi Q1 pro, put a groove, or a lump, or a concave wall in the side of the door so that people can actually open the thing without a tool(I had to jam a butter knife into it earlier).. I understand that this isn't a software update sorta fix, but it should be implemented for those that buy this printer in the future.

bitbooster, 2025-04-27:
Hello,
I'm on Linux Mint 22.1 (so, similar to Ubuntu 24.04) and the workaround of changing environment variables 
`WEBKIT_DISABLE_COMPOSITING_MODE=0 WEBKIT_DISABLE_DMABUF_RENDERER=0`
didn't help for me, QIDIstudio still crashes upon connecting to the 3d printer.

The only workaround that worked was to set the port 7125 in print_host field, as per comment
https://github.com/QIDITECH/QIDIStudio/issues/55#issuecomment-2649726195
so at least it's possible to send directly gcode to the printer, but then a direct web browser session to the printer is needed to start the print.
Looking forward for a complete resolution of this issue.

solazs, 2026-01-11:
Latest version seems to work on arch linux setting the recommended envvars.

Unfortunately my new qidi box is not recognized (by the printer), but that's another issue.