QIDITECH/QIDIStudio #9: Cannot run flow calibration (Course): v1.9.5.51 MacOS_ARM64

Xorlent, 2024-09-27:
Please see attached screenshot.

Selecting Coarse calibration and clicking "Calibration" gives error: "Please input valid values: 0.0 <= flow ratio < 2"

- Plus 4 with stock .4mm nozzle
- Generic PETG filament profile.  Flow ratio is currently .955
- 0.20mm Standard process

Thank you
[image]

QIDITECH, 2024-09-28:
OK, we will fix it in next version.

Xorlent, 2024-09-28:
In the meantime, if any users run into bugs that prevent full functionality, I've ported the Qidi Studio X Plus 4 printer profiles to OrcaSlicer:

[https://github.com/Xorlent/Orcaslicer-Qidi-Plus-4](https://github.com/Xorlent/Orcaslicer-Qidi-Plus-4)

QIDITECH, 2024-09-29:
Thank you very much for your sharing, we have uploaded the configuration to orca, and X-Plus4 configuration has been included in the latest V2.2.0-beta2.

Xorlent, 2024-09-29:
> Thank you very much for your sharing, we have uploaded the configuration to orca, and X-Plus4 configuration has been included in the latest V2.2.0-beta2.

Great news, thank you!

QIDITECH, 2024-09-30:
This bug has been fixed in the latest release.

closed 2024-09-30 by QIDITECH