QIDITECH/QIDIStudio #11: Autoupdate Downloads Wrong CPU Architecture

aidan-gibson, 2024-09-29:
I got the "new version of qidi studio" notification, hit download, and the "Intel" / x86 version was the version downloaded. The version I already had was the arm x64 version (I have an m-series mac / apple "silicon"). The update offer should offer the correct CPU architecture.

The workaround, of course, is manually downloading the correct version from releases on this repo. 

Thanks.

QIDITECH, 2024-09-30:
In general, arm64 systems are compatible with x86 versions of software.
However, we will optimize this problem in the future.

aidan-gibson, 2024-09-30:
Yep, it'll "work", but seems silly to direct users updating to the wrong CPU arch when you have both options available. I'll see if I can find the time for a PR to fix this at some point. Alternatively you could just release a universal build for the app, but that'd probably be more work, I'm not sure. Thanks!