QIDITECH/QIDIStudio #12: Pressure advance calibration not printing properly

Qjeezy, 2024-09-30:
When doing the pressure advance calibration, it is printing the tests features in the wrong order. 

It prints the first layer box that it puts the numbers on, then it prints the numbers on layer 2, and then it tries to print the actual test lines back on layer 1. 

The test lines fail to stick to the build plate. I can only assume because it is trying to print these lines on layer 2 when there is nothing on layer 1. 

The print needs to complete the test lines before it moves onto printing the numbers.

QIDITECH, 2024-10-04:
This is not a problem of printing order, but because of the problem of area leveling, because the calibration procedure only takes the left digital area as the leveling area, which leads to an error in the print distance of the right underlined area.
We will fix this issue in a later version.

Qjeezy, 2024-10-04:
Awesome! Looking forward to a fix. Thanks

QIDITECH, 2024-11-06:
We have fixed the problem in the latest version, please update the software.