QIDITECH/QIDIStudio #13: Z-Offset Feature?

Kr3tt3k, 2024-09-30:
Hi,

is there any feature to set a Z-Offset for each print.
Can't find it in the settings, but would be nice as other slice have it.

Kr3tt3k, 2024-10-04:
Thanks for your reply.
Will it be added in future versions then?+-+-

QIDITECH, 2024-10-05:
With the development of automatic leveling technology, the use of this function is less and less, and it is not suitable for beginners. I don't plan to add this parameter for now, but I will gather feedback from other users and make a decision in a future release.

RobWickham, 2025-01-14:
I would like this capability to be modified from how I currently see it in the Plus4. I'm finding that it only has a responsive UI once the print has begun. I would like to be able to set the z offset any time -- I do not like feeling rushed to do it in the very few seconds before my model prints and I need it to be slightly different for Nylon vs. TPU, vs. ABS. My old printer would allow this -- a heated chamber and soaked hotbed are not a panacea. Also, an easier way to get it back to zero please.

Mulesii, 2025-01-19:
Auto-levelling is great, but I still need to add a Z Offset of .035 to my Qidi Plus4 for PLA and PETG filaments in order to get a good first layer.   I currently add it to the start gcode for each filament, but would be nice if it was a setting.  I disagree that this is not suitable for beginners.

nZambi-, 2025-01-19:
I'd also like to see this returned.  "not suitable for beginners" is probably not the best way to decide which features stay and which do not. You can Hide it behind "Advanced" toggle...  Z-Offset is probably not overly advanced for less experienced printer operators.

Kr3tt3k, 2025-01-26:
I would like to speak out again for adding a z-offset feature as me too has still to handle with different kinds of filaments which need to have different z-offsets for initial layers.
Would be very useful to have this option like in Qidi Slicer!

compf99, 2025-02-11:
z-offset is something "that should be"

compf99, 2025-07-17:
I finally solved the Z offset problem with these commands in printer.cfg at end. 
#*# [smart_effector]
#*# z_offset = -0.058
Yours may have other values, depending on how it behaves on the first layer, mine was too thin. I hope this information helps.

anatol, 2026-04-25:
Here is my attempt to implement the feature https://github.com/QIDITECH/QIDIStudio/pull/186/changes/128c3ed72b3ba8a09953d2595c0acb73a0c2b931 Please take a look