QIDITECH/QIDIStudio #17: Forgot to

EsQueue81, 2024-10-06:
Simple functionality features in Orca that are missing from Qidi Studio. 

1) The collapse sidebar button is missing is this program. In Orca, it is right next to the upper portion of the sidebar. It is an Icon that has a left and right pointing Chevron arrow. With the mouse pointing to it is says "Collapse Sidebar (Shift + Tab)" if opened and "Expand Sidebar (Shift + Tab)" if closed. 

2) When the program is set in windowed mode, the minimize, Full-screen and Close buttons all disappear. These all work like all programs do in Orca. If in windowed mode, you must double-click the top to go back into full screen and while in windowed mode, there isn't an option to minimize the screen without first going back to full screen. 

Bad and extremely annoying feature in both programs is when in full screen, they both hide your autohid taskbar in windows. Only way to access taskbar is to hit the Windows key on the keyboard or put it into windowed mode. 

The first 2 issues should be incredibly easy to fix and are problems especially as they are working in Orca Slicer. The last I hope it get fixed.

QIDITECH, 2024-10-07:
Sorry, we cannot reproduce question 2, please tell me your computer's brand, configuration, system version, etc., so that we can reproduce the problem.

EsQueue81, 2024-10-07:
Thanks for the quick reply. Thanks for the quick reply. Now that I reloaded the program, it seems to keep the window icons without problems when not if full screen mode anymore. Since that was my first time using the program I assumed that this was the programs fault as I've never seen it before but It seems that I just needed to close and restart the program after installing it. 

You can ignore problem 2. Thanks again.

EsQueue81, 2024-10-07:
Here is my system info if it helps with the other problems. 
OS Name	Microsoft Windows 11 Pro
Version	10.0.22635 Build 22635
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	ANTEC_V1
System Manufacturer	ASRock
System Model	Z790 Steel Legend WiFi
System Type	x64-based PC
System SKU	To Be Filled By O.E.M. (Self Built)
Processor	13th Gen Intel(R) Core(TM) i7-13700KF, 3400 Mhz, 16 Core(s), 24 Logical Processor(s) 
BIOS Version/Date	American Megatrends International, LLC. 17.01, 8/8/2024
SMBIOS Version	3.5
Embedded Controller Version	255.255
BIOS Mode	UEFI
BaseBoard Manufacturer	ASRock
BaseBoard Product	Z790 Steel Legend WiFi
BaseBoard Version	
Platform Role	Desktop
Secure Boot State	Off
PCR7 Configuration	Elevation Required to View
Windows Directory	C:\Windows
System Directory	C:\Windows\system32
Boot Device	\Device\HarddiskVolume6
Locale	United States
Hardware Abstraction Layer	Version = "10.0.22621.2483"
User Name	XXXXXXXXXXX
Time Zone	Pacific Daylight Time
Installed Physical Memory (RAM)	32.0 GB
Total Physical Memory	31.8 GB
Available Physical Memory	11.9 GB
Total Virtual Memory	61.8 GB
Available Virtual Memory	21.9 GB
Page File Space	30.0 GB
Page File	C:\pagefile.sys
Kernel DMA Protection	On
Virtualization-based security	Running
Virtualization-based security Required Security Properties	
Virtualization-based security Available Security Properties	Base Virtualization Support, DMA Protection, UEFI Code Readonly, SMM Security Mitigations 1.0, Mode Based Execution Control, APIC Virtualization
Virtualization-based security Services Configured	Hypervisor enforced Code Integrity
Virtualization-based security Services Running	Credential Guard, Hypervisor enforced Code Integrity
Windows Defender Application Control policy	Enforced
Windows Defender Application Control user mode policy	Off
Device Encryption Support	Elevation Required to View
A hypervisor has been detected. Features required for Hyper-V will not be displayed.

nZambi-, 2025-01-06:
This is just a note to say that the inability to collapse the sidebar is also a fairly big inconvenience to me as well. 

I'm able to reproduce the error num2 above #https://github.com/QIDITECH/QIDIStudio/issues/17#issue-2568503588 
Windows 10
QIDI Studio 1.10.1.5

"Restore-Down" window and resize it to minimum width.
Close app.
Open app.
minimize, restore and maximize buttons missing. 
If you resize by dragging they are restored, but if maximized, then restored they are not. 
[image]