QIDITECH/QIDIStudio #18: new feature request

rednet1, 2024-10-08:
it would be possible at the "send to print" moment to add a flag to enable or disable the kamp mesh, sometimes repeating it makes no sense.

regards

QIDITECH, 2024-10-09:
Ok, so we'll think about how do we add this feature

rednet1, 2024-10-09:
[image]

:)

compf99, 2025-02-12:
"enable or disable the kamp mesh" ia a "must be". 
Very useful option if you want the same model again but ... without peeling the old model by bending. If you leave it to cool until the old model peels off naturally then I think this option must exist. It is a very useful option, you must implement it! Thank you!

rednet1, 2025-02-14:
This function is already present in the printer display, for an object already printed but it is inconvenient to use.

compf99, 2025-07-23:
> This function is already present in the printer display, for an object already printed but it is inconvenient to use.

Why, you can activate kamp before send to print. Thank you QIDI Team, you save a lot of time for me !