QIDITECH/QIDIStudio #19: QidiStudio 1.9.7.52 Crash Lib et GTK on Linux (Fedora40)

denisbkm, 2024-10-08:
I'm having a problem with QidiStudio 1.9.7.52 and it crashes when the AppImage is launched.
There seems to be a problem with your custom Lib ‘libqidi_networking.so’ and a repercussion on the GTK Libs.

**Here is my launch log:**

[denisxxx@xxx Software]$ ./QIDIStudio_1.9.7.52_Linux.AppImage
[2024-10-08 15:59:44.036646] [0x00007f302ea50680] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/denisxxx/.config/QIDIStudio/plugins/libqidi_networking.so
error, dlerror is /home/denisxxx/.config/QIDIStudio/plugins/libqidi_networking.so: Cannot open the shared object file: No files or folders of this name
after dlopen, network_module is (nil)

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:46.961: gtk_window_resize: assertion 'height > 0' failed

(qidi-studio:84268): GLib-GObject-CRITICAL **: 15:59:46.981: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:46.981: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-studio:84268): GLib-GObject-CRITICAL **: 15:59:46.982: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:46.982: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:46.982: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:46.982: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): GLib-GObject-CRITICAL **: 15:59:46.982: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:46.982: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:46.983: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:46.983: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): GLib-GObject-CRITICAL **: 15:59:46.983: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:46.983: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-studio:84268): GLib-GObject-CRITICAL **: 15:59:46.986: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:46.986: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-studio:84268): GLib-GObject-CRITICAL **: 15:59:46.986: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:46.986: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (qidi-studio:84268): CRITICAL **: 15:59:47.127: Cannot register URI scheme wxfs more than once

** (qidi-studio:84268): CRITICAL **: 15:59:47.127: Cannot register URI scheme memory more than once

** (qidi-studio:84268): CRITICAL **: 15:59:47.176: Cannot register URI scheme wxfs more than once

** (qidi-studio:84268): CRITICAL **: 15:59:47.176: Cannot register URI scheme memory more than once

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.239: gtk_window_resize: assertion 'height > 0' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.241: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.241: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.241: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.241: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.241: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.241: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.241: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.241: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.245: gtk_window_resize: assertion 'height > 0' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.245: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.245: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.300: gtk_window_resize: assertion 'height > 0' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.305: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.305: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.305: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.305: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.310: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.310: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.310: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.310: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.312: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.312: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.312: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.312: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.315: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.315: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.315: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.315: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.318: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.318: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.318: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.318: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.321: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.321: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.321: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.321: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.325: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.325: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.325: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:84268): Gtk-CRITICAL **: 15:59:47.325: gtk_widget_set_size_request: assertion 'width >= -1' failed

** (qidi-studio:84268): CRITICAL **: 15:59:47.334: Cannot register URI scheme wxfs more than once

** (qidi-studio:84268): CRITICAL **: 15:59:47.334: Cannot register URI scheme memory more than once
terminate called without an active exception
Abandon (core dumped)

QIDITECH, 2024-10-09:
Thanks for your feedback, fedora and ubuntu are quite different and cannot be compatible at the same time, we will develop a compatible version of fedora in the subsequent version. Please use another system to run the software for the time being.

denisbkm, 2024-10-09:
Hello,
just for information, even having updated the webkit2gtk 4.1 and webkittgtk 6.0 packages, the error is still present, but with a small launch and I put you a video of the problem: https://drive.proton.me/urls/AZS6S2FTQR#OpNesPlL4fdC

I think that for the next version of QidiStudio, it would be preferable that you use the Flatpack => https://flatpak.org system for better compatibility management between all Linux OS (Bambou Studio is present in the FlatHub => https://flathub.org/).
This will also allow all Linux systems to automatically detect the new version when upgrading :o)
Best regards
Denis

QIDITECH, 2024-10-09:
Thank you. We'll fix it as soon as possible.

denisbkm, 2024-10-20:
Hello,
just for information with Orca Slicer (Appimage) on Linux Fedora, PopOS, Ubuntu :
1) Using version 2.1.1 Release it works, but crashes when you want to send the print to the printer.
2) Using version 2.2.0RC everything works correctly!
It seems that perhaps the printer connection plugin is causing problems :o(
See if this can help you for your next Qidi Studio update ;o)
Best regards
Denis

denisbkm, 2024-11-26:
Hello Qiditech,
By launching  new QIDIStudio_1.9.7.53_Linux.AppImage still the crash problem after startup on Fedora 41, PopOS with a Gtk-CRITICAL and your libqidi_networking module

return log:

$ ./QIDIStudio_1.9.7.53_Linux.AppImage
[2024-11-26 18:03:28.586328] [0x000072f574df6400] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:29.063: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
loading network module at /home/denisbp/.config/QIDIStudio/plugins/libqidi_networking.so
error, dlerror is /home/denisbp/.config/QIDIStudio/plugins/libqidi_networking.so: Ne peut ouvrir le fichier d'objet partagé: Aucun fichier ou dossier de ce nom
after dlopen, network_module is (nil)

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.605: gtk_window_resize: assertion 'height > 0' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.670: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.670: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): GLib-GObject-WARNING **: 18:03:48.670: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.670: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.671: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.671: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): GLib-GObject-WARNING **: 18:03:48.671: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.671: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.673: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.673: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): GLib-GObject-WARNING **: 18:03:48.673: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.673: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.674: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.674: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): GLib-GObject-WARNING **: 18:03:48.674: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.674: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-studio:5873): GLib-GObject-WARNING **: 18:03:48.678: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.678: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-studio:5873): GLib-GObject-WARNING **: 18:03:48.679: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.679: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (qidi-studio:5873): CRITICAL **: 18:03:48.849: Cannot register URI scheme wxfs more than once

** (qidi-studio:5873): CRITICAL **: 18:03:48.849: Cannot register URI scheme memory more than once

** (qidi-studio:5873): CRITICAL **: 18:03:48.853: Cannot register URI scheme wxfs more than once

** (qidi-studio:5873): CRITICAL **: 18:03:48.853: Cannot register URI scheme memory more than once

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.881: gtk_window_resize: assertion 'height > 0' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.883: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:48.883: gtk_widget_set_size_request: assertion 'width >= -1' failed
[... 38 lines cut ...]

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:49.068: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:49.075: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:49.075: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:49.081: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:49.081: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:49.088: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-studio:5873): Gtk-CRITICAL **: 18:03:49.088: gtk_widget_set_size_request: assertion 'width >= -1' failed

** (qidi-studio:5873): CRITICAL **: 18:03:49.113: Cannot register URI scheme wxfs more than once

** (qidi-studio:5873): CRITICAL **: 18:03:49.113: Cannot register URI scheme memory more than once
terminate called without an active exception
Abandon (core dumped)
---

I think you should switch to Flatpack to avoid the recurring GTK problems caused by AppImage.
Best regards
Denis

wjyLearn, 2025-03-10:
Can temporarily refer to #8