QIDITECH/QIDIStudio #23: Segmentation fault (core dumped)

y-e-r, 2024-10-13:
Connecting to printer over LAN results in Segmentation fault (core dumped)
Simply going to the "Device" tab and clicking on the printer causes an immediate crash.

Fedora 40, KDE desktop
webkit2gtk4.0-2.44.3-2.fc40.x86_64
QIDIStudio 1.9.7.52 appimage
X-Plus 3

QIDITECH, 2024-10-14:
Thanks for your feedback, fedora and ubuntu are quite different and cannot be compatible at the same time, we will develop a compatible version of fedora in the subsequent version. Please use another system to run the software for the time being.

snigel, 2025-02-11:
@y-e-r is it the same behavior in 1.10.01.51 as well? Thinking whether this is related to #55

wjyLearn, 2025-03-10:
Can temporarily refer to #8