QIDITECH/QIDIStudio #24: Feature request - Orca Calibrations

WyldCFH, 2024-10-15:
Hello,

Please add the full calibrations from Orca into Qidi Studio.

[image]

QIDITECH, 2024-10-16:
Orca Calibrations have many bugs, and it is easy for beginners to misoperate.
We will choose the calibration method suitable for our printer after testing, and make a detailed introduction and guidance.

dssamusaran, 2024-12-24:
I do second this idea.

Even if it's not perfect in Orca, Temperature Calibration, as well as retraction calibration is required to tune up filaments.
Your machine works great but people need those tools to be able to use it at it's full potential with other brands of filaments.

Until now I'm tediously copying over the QidiStudio profiles to Orca just to be able to use the other calibration types there. And then tuning my QidiStudio profiles.

On the other side, I do really like the clear as water guidance and the efforts you put into the flow rate/pressure advance/volumetric speed that are available. Great work! We only need more of those. :)