QIDITECH/QIDIStudio #28: curl: Couldn't connect to server

ethanholter, 2024-10-21:
using QIDI Q1 Pro

when attempting to send gcode using the "Print plate" button after slicing I get the following error message

```
Error uploading to print host:
curl:Couldn't connect to server:
Failed to connect to 192.168.1.209 port 80: Connection refused
[Error 7]
```

looks like the default port that QIDIStudio uses to connect with the printer is incorrect. Not super knowledgeable about the software used but it appears to be attempting to connect to moonraker (?). I attempted changing the default port for moonraker using the moonraker.conf file but was getting a read-only forbidden error.

Below is the nmap scan of my printer. 

```
PORT      STATE SERVICE
22/tcp    open  ssh
7125/tcp  open  unknown
8080/tcp  open  http-proxy
8990/tcp  open  http-wmap
10088/tcp open  unknown
```

QIDITECH, 2024-10-21:
Please try adding :10088 after IP

ethanholter, 2024-10-22:
> Please try adding :10088 after IP

Worked for me. You guys can decide whether to close the issue. Not sure if this is intended behavior since everything else works fine without manually specifying the port

QIDITECH, 2024-10-23:
Normally the port number is automatically added to the IP :10088, I'm not sure why it doesn't work on your computer. In order to better reproduce the problem, please tell me your printer firmware version, software version, computer system version.