QIDITECH/QIDIStudio #31: Export Preset Bundle - Not able to export Filament Presets

67RS427, 2024-11-09:
Can't export user filament presets...

macOS Sonoma v14.7
Qidi Studio v1.9.7.53

File / Export / Export Preset Bundle:
*Filament Preset Bundle - Does not show any of the User filament presets
[image]
[image]

File / Export / Export Preset Bundle:
*Filament Presets (.zip) - Does not show anything at all...
[image]

QIDITECH, 2024-11-11:
Ok, we will fix this issue in the next release.

wjyLearn, 2025-02-05:
The issue has been addressed in version 1.9.7.53, therefore this issue will be closed.

closed 2025-02-05 by wjyLearn