QIDITECH/QIDIStudio #32: Temperature Calibration Tower Missing

67RS427, 2024-11-09:
macOS Sonoma v14.7
Qidi Studio v1.9.7.53

Right Click QidiStudio.app, Show package contents - /Applications/3D PRINTING/QIDIStudio.app/Contents/resources/calib/temperature_tower/temperature_tower.stl

The temp tower STL is shown, but not available in the app. The other calibration options should be shown as well.

QIDITECH, 2024-11-11:
#24
Orca Calibrations have many bugs, and it is easy for beginners to misoperate.
We will choose the calibration method suitable for our printer after testing, and make a detailed introduction and guidance.

67RS427, 2024-11-11:
I understand but a temperature tower calibration is very important. Please consider fixing the STL and adding it to your calibration menu. Thanks!

dssamusaran, 2024-12-24:
I do second this idea.

Even if it's not perfect in Orca, Temperature Calibration, as well as retraction calibration is required to tune up filaments.
Your machine works great but people need those tools to be able to use it at it's full potential with other brands of filaments.

Until now I'm tediously copying over the QidiStudio profiles to Orca just to be able to use the other calibration types there. And then tuning my QidiStudio profiles.

On the other side, I do really like the clear as water guidance and the efforts you put into the flow rate/pressure advance/volumetric speed that are available. Great work! We only need more of those. :)

dssamusaran, 2025-07-02:
> This is honestly the worst response I've ever seen they don't give a reason as to what makes it so terrible they just say it's bad so we don't use it meanwhile every other slicer has no problem getting it working

Exactly! A temp calibration method is the first step with a new filament and at Qidi they removed it. At the pace it's going we'll never have it back.

If it helps, meanwhile I'm using this script and procedure: https://www.printables.com/model/151914-ultimate-customizable-temperature-tower

You get some stl, slice it and export the gcode. Then use the script to add the temp changes to the gcode.
You also have an openscad version that allows to do a fully customized to your need temp tower stl.

Rudokhvist, 2025-07-02:
> Exactly! A temp calibration method is the first step with a new filament and at Qidi they removed it. 

They didn't remove it. QIDIStudio is based on BambuStudio (that don't have a temperature tower calibration), and not on Orca.

dssamusaran, 2025-07-02:
> > Exactly! A temp calibration method is the first step with a new filament and at Qidi they removed it.
> 
> They didn't remove it. QIDIStudio is based on BambuStudio (that don't have a temperature tower calibration), and not on Orca.

Good point. I always thought it was based on orca rather than Bambu studio.

Still, both of them should implement a temp tower or any other system to calibrate the temperature if they feel there is a better way.

bgswiss1, 2025-07-03:
> > Exactly! A temp calibration method is the first step with a new filament and at Qidi they removed it.
> 
> They didn't remove it. QIDIStudio is based on BambuStudio (that don't have a temperature tower calibration), and not on Orca.

I looked it up and you're right but that makes it slightly worse because when looking it up it said orca slicer is based off of bambu. orca slicer has so many calibrations that it makes this an even worse argument.(if another slicer using the same base code as you is able to carry the entire calibration suite why does qidi only give us three)