QIDITECH/QIDIStudio #33: Anwendung auf Ubuntu 24.10 nicht möglich.

santahx, 2024-11-17:
When attempting to run QIDISlicer_1.1.7_Linux.AppImage, the following error is encountered:

error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

Efforts to resolve this by installing the missing library or creating symbolic links have failed due to the following:

    Missing libwebkit2gtk-4.0.so.37:
        The package libwebkit2gtk-4.0-37 or libwebkit2gtk-4.0.so.37 is not available in the repository.
        Symbolic linking libwebkit2gtk-4.0.so.38 to libwebkit2gtk-4.0.so.37 does not work because libwebkit2gtk-4.0.so.38 itself is not found in /usr/lib/x86_64-linux-gnu/.

    Relevant Installed Packages:
        libwebkit2gtk-4.1-0 and libwebkit2gtk-4.1-dev are installed (version 2.46.1-0ubuntu1).
        libwebkit2gtk-4.0-doc is installed but does not provide the missing .so file.

    Steps Taken:
        Attempted to install libwebkit2gtk-4.0.so.37 or libwebkit2gtk-4.0-37 but could not locate these packages.
        Created a symbolic link:
sudo ln -s /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.38 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
However, this failed since libwebkit2gtk-4.0.so.38 does not exist either.
System Details:

    Ubuntu version: Oracular (Ubuntu 24.10  or newer).
    Installed libwebkit2gtk version: 2.46.1-0ubuntu1.
--------
Steps to Reproduce

    Download the QIDISlicer_1.1.7_Linux.AppImage.
    Make it executable: chmod +x QIDISlicer_1.1.7_Linux.AppImage
    Run the application: ./QIDISlicer_1.1.7_Linux.AppImage

------------------------
Here is the GitHub issue description in English:
Description of the Issue

When attempting to run QIDISlicer_1.1.7_Linux.AppImage, the following error is encountered:

error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

Efforts to resolve this by installing the missing library or creating symbolic links have failed due to the following:

    Missing libwebkit2gtk-4.0.so.37:
        The package libwebkit2gtk-4.0-37 or libwebkit2gtk-4.0.so.37 is not available in the repository.
        Symbolic linking libwebkit2gtk-4.0.so.38 to libwebkit2gtk-4.0.so.37 does not work because libwebkit2gtk-4.0.so.38 itself is not found in /usr/lib/x86_64-linux-gnu/.

    Relevant Installed Packages:
        libwebkit2gtk-4.1-0 and libwebkit2gtk-4.1-dev are installed (version 2.46.1-0ubuntu1).
        libwebkit2gtk-4.0-doc is installed but does not provide the missing .so file.

    Steps Taken:
        Attempted to install libwebkit2gtk-4.0.so.37 or libwebkit2gtk-4.0-37 but could not locate these packages.
        Created a symbolic link:

        sudo ln -s /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.38 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37

        However, this failed since libwebkit2gtk-4.0.so.38 does not exist either.

    System Details:
        Ubuntu version: Oracular (Ubuntu 23.10 or newer).
        Installed libwebkit2gtk version: 2.46.1-0ubuntu1.

Expected Behavior

The application should run successfully without errors regarding missing libwebkit2gtk-4.0.so.37.
Actual Behavior

The application fails to launch with an error indicating that libwebkit2gtk-4.0.so.37 cannot be found.
Steps to Reproduce

    Download the QIDISlicer_1.1.7_Linux.AppImage.
    Make it executable:

chmod +x QIDISlicer_1.1.7_Linux.AppImage

Run the application:

    ./QIDISlicer_1.1.7_Linux.AppImage

Additional Notes

The dependency issue seems to stem from the application's reliance on an outdated version of libwebkit2gtk (4.0) that is no longer available in the repositories. Current systems include only libwebkit2gtk-4.1.

Workaround Attempts:

    Symbolic links to libwebkit2gtk-4.0.so.37 failed as the corresponding .so.38 file does not exist.
    Installing newer versions (4.1) does not resolve the problem.

Suggestions

    Update the AppImage to use libwebkit2gtk-4.1 instead of libwebkit2gtk-4.0.
    Include the required .so files within the AppImage itself to avoid dependency conflicts.

santahx, 2024-11-17:
Orca-Slicer was facing the same issue
it was solved by setting the newer version to be checked, temporarly it was solved by creating a link:
sudo ln -s /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.38 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
but it worked not for me in this case.
https://github.com/SoftFever/OrcaSlicer/issues/4616

https://github.com/SoftFever/OrcaSlicer/issues/5860

QIDITECH, 2024-11-18:
We will release libwebkit2gtk-4.1-0 compatible version in the next version, please wait for the next update.

santahx, 2024-11-19:
Thanks for The fast response, i would Like to add arc Supports or is it currently in planning?

QIDITECH, 2024-11-20:
What do you mean arc Supports? There are a variety of related functions I can relate to, not sure which one you are referring to.

santahx, 2024-11-20:
https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration

Sorry for my previous miss spelling

QIDITECH, 2024-11-21:
Unfortunately, this feature is not currently in our plans. According to our test results, the practicality and stability of this function are not perfect.

logiclinux, 2024-11-23:
Hi facing the same problem i want et to compile Qidi Studio  from the .targz file but struggle to get the path for the cereal
 cmake -Dcereal_DIR=/path/to/cereal ..
didn't help