QIDITECH/QIDIStudio #34: Network connection

janzwah, 2024-11-19:
unable to connect via the AWS or Aliyun server to my Fluidd page.

QIDITECH, 2024-11-20:
Can you provide videos or pictures to describe the problem in more detail? Which specific step went wrong?

janzwah, 2024-11-21:
[image]

The problem is, in the QIDI Studio in the linked Printer tap I only see the screen I showed in the Picture. In the QIDI Link app everything is working.

I tried both Server AWS and Aliyun, currently I use the AWS.

QIDITECH, 2024-11-22:
The page is not redirected, it has nothing to do with the network connection, please tell us your computer's system and configuration, we will try to replicate and resolve the problem.

janzwah, 2024-11-22:
i have a MacBook Pro with a M1 Pro processor. Currently I'm having the macOS Sequoia 15.1.1. Let me know if you need anything else.

closed 2024-12-08 by janzwah