QIDITECH/QIDIStudio #37: Qidi Studio crashing in Windows 11

dsiffred, 2024-11-27:
Qidi studio keeps crashing in windows 11 every time . Qidi studio version is 1.9.7.53 , windows 11 version is 24H2 (Os Build 26100.2454).
Attaching dump file from Qidi software 
[crash_Mon_Nov_25_17_30_18_0_done.log](https://github.com/user-attachments/files/17913182/crash_Mon_Nov_25_17_30_18_0_done.log)

please help, I'd love to solve this issue so I can use my new 3d printer.
thanks!

QIDITECH, 2024-11-28:
Random? You said he would break down every time.
So do you remember what you used to do before you broke down?

craftyham, 2024-12-16:
I am having a similar issue trying to load the updated version . If yes is clicked it closes the app the same happens clicking no. This issue happens load the app.  
Edition	Windows 11 Pro
Version	24H2
Installed on	‎12/‎6/‎2024
OS build	26100.2605
Experience	Windows Feature Experience Pack 1000.26100.36.0

[image]

[image]

QIDITECH, 2024-12-17:
Sorry, the problem cannot be corrected, please provide your 3mf or stl.

craftyham, 2024-12-17:
The issue happens while trying to load Studio. Studio does not open after
the update was installed. There is not a 3mf or stl file involved in this
issue.

QIDITECH, 2024-12-18:
Please go to this path and send the configuration folder to us. AppDate is usually a hidden folder. Show the hidden folder.
[image]
After sending the configuration folder, you can delete the entire folder so that you can continue to open the software normally.

dsiffred, 2024-12-19:
image.png (view on web)<https://github.com/user-attachments/assets/497d05f1-6143-4494-83b6-42945e8d1d93>

You are receiving this email because you authored the thread.

timyoung75, 2024-12-29:
I'm also having the same issue on Windows 11. Previous version was 1.9.7.x was working fine until yesterday but fails to load. 

Tried uninstalling, installing the new version (1.10.01.50) and clearing the Roaming data folder as mentioned above.

The error being reported is 

Faulting application name: qidi-studio.exe, version: 1.10.1.50, time stamp: 0x675bf13d
Faulting module name: QIDIStudio.dll, version: 0.0.0.0, time stamp: 0x675bf138
Exception code: 0xc000041d
Fault offset: 0x0000000000cf5fa9
Faulting process id: 0x0x61B0
Faulting application start time: 0x0x1DB5987E2168957
Faulting application path: C:\Program Files\QIDIStudio\qidi-studio.exe
Faulting module path: C:\Program Files\QIDIStudio\QIDIStudio.dll
Report Id: b23ed136-c8ef-42ba-b3b5-4805ae4a91c1
Faulting package full name: 
Faulting package-relative application ID:

TinkerBarn, 2025-04-13:
I have also random chrashing of QIDI Studio 1.10.2.76 in WIndows 11.
There is no error message. It suddenly disapears if the window is in background and I have any other application open, like Chrome browser or YouTube or InksCape or, or, or....

I only see, oh the icon for the open QIDI STudio session is no longer showing in the status bar of WIndows 11.

If I again open Qidi Studio I get always the question, if i would restore the last project.

I found out, that QIDI Studio always creates a crash log in C:\Users\<user>\AppData\Roaming\QIDIStudio\log
For example: crash_Fri_Apr_04_00_09_14_0_done.log

Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x308 
Context :0000003B930CA980 
ContextFlag : 0x10005f, EFlags: 0x10246 
Fault address:  0x23D2F509 0x0:0x0 

Show CallStack:
Call SymGetLineFromAddr64 ,Address 23d2f509 Error:000001e7
Call SymGetLineFromAddr64 ,Address 23bb737a Error:000001e7
Call SymGetLineFromAddr64 ,Address 23bb7229 Error:000001e7
Call SymGetLineFromAddr64 ,Address 23f656a2 Error:000001e7
Call SymGetLineFromAddr64 ,Address 23ed0abc Error:000001e7
Call SymGetLineFromAddr64 ,Address 23ec7cb3 Error:000001e7
Call SymGetLineFromAddr64 ,Address 24f9e732 Error:000001e7
Call SymGetLineFromAddr64 ,Address 24f962d4 Error:000001e7
Call SymGetLineFromAddr64 ,Address 24f9669a Error:000001e7
Call SymGetLineFromAddr64 ,Address 24f96963 Error:000001e7
Call SymGetLineFromAddr64 ,Address 24f961ea Error:000001e7
Call SymGetLineFromAddr64 ,Address 24f9660a Error:000001e7
Call SymGetLineFromAddr64 ,Address 24d89f41 Error:000001e7
Call SymGetLineFromAddr64 ,Address 24d9eff2 Error:000001e7
Call SymGetLineFromAddr64 ,Address 24da246a Error:000001e7
Call SymGetLineFromAddr64 ,Address 24dee1c6 Error:000001e7
Call SymGetLineFromAddr64 ,Address 23f52b6c Error:000001e7
Call SymGetLineFromAddr64 ,Address 24d8ce6c Error:000001e7
Call SymGetLineFromAddr64 ,Address db36b643 Error:000001e7
Call SymGetLineFromAddr64 ,Address db36ad1c Error:000001e7
Call SymGetLineFromAddr64 ,Address db3b5706 Error:000001e7
Call SymGetLineFromAddr64 ,Address dc203814 Error:000001e7
Call SymGetLineFromAddr64 ,Address d9a712b4 Error:000001e7
Call SymGetLineFromAddr64 ,Address db37436f Error:000001e7
Call SymGetLineFromAddr64 ,Address db3742f8 Error:000001e7
Call SymGetLineFromAddr64 ,Address 24fcf68f Error:000001e7
Call SymGetLineFromAddr64 ,Address 24fcef43 Error:000001e7
Call SymGetLineFromAddr64 ,Address 24fcf1da Error:000001e7
Call SymGetLineFromAddr64 ,Address 24f9f6d7 Error:000001e7
Call SymGetLineFromAddr64 ,Address 24fa1a27 Error:000001e7
Call SymGetLineFromAddr64 ,Address 2500bfc8 Error:000001e7
Call SymGetLineFromAddr64 ,Address 23c8a624 Error:000001e7
Call SymGetLineFromAddr64 ,Address 2339d92b Error:000001e7
Call SymGetLineFromAddr64 ,Address 2340bdd5 Error:000001e7
Call SymGetSymFromAddr64 ,Address 1ef85635 Error:000001e7
Call SymGetSymFromAddr64 ,Address 1ef85af6 Error:000001e7
Call SymGetLineFromAddr64 ,Address db77e8d7 Error:000001e7
Call SymGetLineFromAddr64 ,Address dc15bf6c Error:000001e7

wjyLearn, 2025-04-14:
You can try to delete the configuration file and then start the software. Before deleting it, we hope you can package and upload the configuration file so that we can better reproduce the problem.