QIDITECH/QIDIStudio #38: Packaging Request Flatpak for Linux

ZekeZDev, 2024-12-06:
Would be really nice to see a release on Flathub and packaging as flatpak, this would help with most dependency issues and show in software centers for linux users. 

Existing slicers and ones this is based on bambu and prusa already have flatpak packages on flathub and it makes installing them incredibly easy. There is a couple edge cases with flatpaks but I think overall they are incredibly convenient and work much better across distros.

Other slicers pages on flathub:
https://flathub.org/apps/com.bambulab.BambuStudio
https://flathub.org/apps/com.prusa3d.PrusaSlicer

[image]
Example in my software center by default (fedora, popos, linux mint) mainly ubuntu doesnt bundle flatpak by default but it can be enabled

QIDITECH, 2024-12-07:
Thank you for your suggestions and look forward to the next update.

fl0wfr, 2025-05-01:
We definitely need a way to properly install the app on other distributions than Ubuntu.
BuildLinux.sh fails on Fedora 42, AppImage won't run on Fedora 42...

ZekeZDev, 2025-05-22:
iirc that was related to newer packages and also happens on latest ubuntu too? (source needed)