QIDITECH/QIDIStudio #39: QidiStudio 1.10 fails to boot in Ubuntu 24.04

dan-1d, 2024-12-06:
The recently-released QidiStudio version 1.10 does not seem to run in ubuntu 24.04

```
~/Downloads$ ./QIDIStudio_1.10.01.50_Linux_ubuntu_24.04.AppImage 
fusermount3 version: 3.14.0
/tmp/.mount_QIDIStFdCpjG/bin/qidi-studio: error while loading shared libraries: libavcodec.so.61: cannot open shared object file: No such file or directory
```

Ubuntu 24.04 contains libavcodec version 60, not 61:
```
$ find /usr/ -name "libavcodec*"
/usr/include/x86_64-linux-gnu/libavcodec
/usr/lib/x86_64-linux-gnu/pkgconfig/libavcodec.pc
/usr/lib/x86_64-linux-gnu/vlc/plugins/codec/libavcodec_plugin.so
/usr/lib/x86_64-linux-gnu/libavcodec.a
/usr/lib/x86_64-linux-gnu/libavcodec.so
/usr/lib/x86_64-linux-gnu/libavcodec.so.60.31.102
/usr/lib/x86_64-linux-gnu/libavcodec.so.60
/usr/share/doc/libavcodec-dev
/usr/share/doc/libavcodec-extra60
/usr/share/doc/libavcodec-extra
/usr/share/lintian/overrides/libavcodec-extra60
```

QIDITECH, 2024-12-07:
Thank you for your suggestions and look forward to the next update.

TopheC, 2024-12-14:
> The new Qidi Studio V 1.10 does not run on any ubuntu version, or linux mint. When you try to integrate and run the appimage file you get the error cannot register the file and it deletes it from my downloads.

That's exactly what happened to me to @jetmech43 
I'm on Linux Mint 22 (Wilma, based on Ubuntu 24.04).

I confirmed also what @dan-1d found : 
Ubuntu 24.04 contains libavcodec version 60, not 61:

```
$ find /usr/ -name "libavcodec.so*"
/usr/lib/x86_64-linux-gnu/libavcodec.so
/usr/lib/x86_64-linux-gnu/libavcodec.so.60.31.102
/usr/lib/x86_64-linux-gnu/libavcodec.so.60

```
@QIDITECH Is there a workaround, at this time ?

QIDITECH, 2024-12-16:
We will try to release the next version on Flathub, which should solve this problem.

QIDITECH, 2024-12-17:
I'm very sorry that I made some mistakes. This problem has been fixed and re-uploaded. Please re-download appimage.

closed 2024-12-17 by QIDITECH