QIDITECH/QIDIStudio #41: Top layer flow rate setting missing

dssamusaran, 2024-12-24:
When doing the calibration of the flow rate it is perfectly spot on in the resulting test plate at '0'.

When printing an object everything seems great during the print, up to the top layer that shows waves and rough texture.
[image]

Don't pay attention to the outside perimeter, I cancelled the print. The roughness that you see and waves where the top layers of the bottom of a little round box.

I get this issue on nearly every print that has large flat top layers. And it appears with any top pattern type. It seems worse with archimedian chords and concentric. But even monotonic lines shows roughness and wavyness.

On my previous slicers (PrusaSlicer & OrcaSlicer) I would've been able to change the top layer flow rate to lower it a tiny bit.
On QidiStudio I can't seem to find it.

Please give us back this option. 
Or else, how can I fix the issue in another way?

QIDITECH, 2024-12-25:
Please enable develop mode in the Preferences.
[image]

QIDITECH, 2024-12-25:
The problem is caused by the warping of the bottom surface of the model, and you can also improve it by increasing the adhesion of the bottom surface, reducing the printing speed, and reducing the heat deformation.

dssamusaran, 2024-12-25:
Hi there, thanks for the develop mode, this'll help me a lot!

For the other input I can 100% confirm for this particular case this is not a warping issue.
The bottom stays perfectly flat on the bed and even whan the pei plate is fully cooled down it still sticks a lot!

closed 2024-12-26 by QIDITECH