QIDITECH/QIDIStudio #42: program cannot save any files for any reason

Excrubulent, 2024-12-29:
I've had QIDI Studio about a week at this point, and it's just picked up this issue out of the blue.  When I try to slice, I get this error message:

```
boost::filesystem::create_directory: The system cannot find the path specified [system:3]:
"C:\Users\<username>\AppData\Local\Temp/qidi_model/Mon_Dec_30/00_14_42#22220#7/Metadata"
```

If I try to save the project, I get this error message:

```
Failed to save the project.
Please check whether the folder exists online or if other programs open the project file or if there is
enough disk space.
```

I don't think I changed anything to cause this, I've tried saving on multiple different drives, all of which have plenty of space on them.

I had to reinstall the program to fix it, and I deleted all my config files, but this was very concerning behaviour to simply start happening out of the blue, and I just wanted to report this issue.

QIDITECH, 2024-12-30:
Strange questions. We'll try to repeat them. If you haven't removed the original configuration and provided it to us, it might be easier.

Excrubulent, 2024-12-30:
Sorry, I didn't think to preserve them.

Excrubulent, 2024-12-31:
Okay, it's happened again.  It worked yesterday after reinstalling, perfectly, then the problem happened again today, with absolutely nothing being changed except it's a new day and I restarted my computer.

I have log files, I also have the full contents of my appdata local and roaming QIDIStudio folders zipped up so I can send them to you if you wish.

Deleting the full contents of both folders fixed absolutely nothing.  It appears that only a reinstall will fix it, which maybe means there's a registry issue, but I don't know where to look to deal with that.  I can try that if you tell me where to look.

Excrubulent, 2024-12-31:
Unless you ask for more file, I'll just upload the contents of my log folder, which is here: 
[log.tar.gz](https://github.com/user-attachments/files/18279376/log.tar.gz)

There are two days, the Monday logs are from when it was working, Tuesday is today, when it didn't work.

Excrubulent, 2024-12-31:
Okay, I can confirm that restarting the computer causes the problem, and re-running the installer, without any other steps, fixes the problem while preserving my settings.  So it's gone from being an insurmountable problem to a very strange annoyance, but I would obviously like to see it fixed.

wjyLearn, 2025-01-02:
From the log you provided, it appears that the temporary folder was created without permission when slicing. You can try to start the software as an administrator, and then try slicing, saving files, and so on.