QIDITECH/QIDIStudio #45: Changing filament at specific layer does not work (no pause)

tohoku, 2025-01-04:
To reproduce:

1. Slice the object
2. (In the Preview tab) Add a "change filament at layer" by right clicking the layer menu on the right side of the screen.
3. Send to printer.
4. The printer will pause briefly at the "filament change" and continue to print.

Expected behavior: The printer should pause to allow me to change filament at point 4. 

Looking at the Gcode for the layer, there is no pause.

Bambu Studio seems to have the same issue:
https://forum.bambulab.com/t/how-do-i-make-a-filament-change-by-layer-in-the-slicer/2503/10
https://forum.bambulab.com/t/change-filament-at-layer-height/7954/12

tohoku, 2025-01-04:
Just for the record: Using QidiSlicer, I can confirm that a M0 Gcode command is inserted, allowing me to change filament properly.

QIDITECH, 2025-01-06:
We are very sorry that we cannot reproduce your problem. Could you please provide a video to confirm how you operate it?

Gradbb, 2025-01-07:
Aften edited your Gcode ,please click "Slice" again.

EsQueue81, 2025-01-09:
> To reproduce:
>
> 1. Slice the object
> 2. (In the Preview tab) Add a "change filament at layer" by right clicking the layer menu on the right side of the screen.
> 3. Send to printer.
> 4. The printer will [...]

I don't expect any responses from tech after the first one. What I did in Qidi Studio was insert a pause command and while paused, I changed the filament using the printer's menu. After changing the filament, I clicked the resume button.

tohoku, 2025-01-10:
Thank you, @EsQueue81 and @Gradbb .

For anyone else who is transitioning from Qidi Slicer to Qidi Studio and have a single filament printer:

In the right-hand layer menu, right click and select "Add Pause" instead of "Add Filament Change".

closed 2025-01-10 by tohoku