QIDITECH/QIDIStudio #46: QIDIStudio Crash on device tab (Appimage 2404)

Mitchou, 2025-01-11:
Hi,

I finally receive my Plus4 and try to slice some STL.
Since i've got connected my printer on my network, I try to add it to Qidi Studio.

But when I clic on it after config, I saw the webpage, freeze and finally crash.
No problem on QidiSlicer or Orca Slicer.

I think he can't launch webcam video stream

Hope it will help for better app
Best regards

QIDITECH, 2025-01-13:
Thank you for your feedback, there are still some problems with running linux, and we will try our best to solve them.

Mitchou, 2025-01-13:
Sorry, I don't mention it. 
I use Fedora 41

I'm couting on you, because I miss tree support on QidiSlicer and OrcaSlicer do mesh bed twice at print start.
Hope you will fix QidiStudio on Linux :D

bednarekpiotr, 2025-01-13:
I'm using Ubuntu and I have problem with QIDIStudio since september. I created a ticket but since this time nothing was fixed.

wjyLearn, 2025-01-17:
The software crash when opening Fluidd in Linux is caused by a bug in the WebKit component's compatibility with Linux. In QIDIStudio you can try to add the port number **7125** when adding the machine, so that you can ensure the normal operation of the software when sending GCode, as for the Fluidd page, for the time being, only through the browser to view.

spajd1, 2025-01-17:
I have the same problem with image app on Ubuntu 24.04, they crash after conect to printer on device tab. OrcaSlicer works perfect on device tab.

snigel, 2025-02-11:
Same issue on Arch running QIDIStudio_1.10.01.51_Linux_Ubuntu_24.04.AppImage

nattymcfly5, 2025-03-09:
Same issue Fedora 41 running QIDIStudio_1.10.02.76_Linux_Ubuntu_24

wjyLearn, 2025-03-10:
#8

wjyLearn, 2025-03-10:
Can temporarily refer to [#8](https://github.com/QIDITECH/QIDIStudio/issues/8)