QIDITECH/QIDIStudio #48: Repairing was canceled

sunbowch, 2025-01-15:
Upon trying to repair any .amf file, a popup appears and says that...

sunbowch, 2025-01-15:
Works perfectly with any version of PrusaSlicer...

QIDITECH, 2025-01-16:
Yes, qidistudio is based on prusaslicer.

shiler00, 2025-01-25:
I'm having the same issue after upgrading to 1.10.1.51.
Your answer is no help at all. We all know it's based on prusa slicer.
Why does repair not work on 1.10.1.51, is the question.

wjyLearn, 2025-02-05:
Please re-download and install version 1.10.1.51 and try again if there is a model repair feature.

sunbowch, 2025-02-06:
Now it seems to work... 
However, it not a good practice to update an existing release of the code when you change the sources to solve a bug... A new sub-release allows everybody to know that something changed.
Closing the issue

closed 2025-02-06 by sunbowch