QIDITECH/QIDIStudio #51: Feature request Custom output filename formats

nZambi-, 2025-01-19:
I'd like to see the option to set Custom output filename formats as seen upstream. I realize that QS moved this to the Send dialog but setting a custom output name once is far less inconvenient than renaming as sent for those who have specific needs for organization. Keep the last-chance to edit in the Send Dialog but add the custom format as seen in Prusa/Orca... please.

[image]

QIDITECH, 2025-01-23:
OK，we will optimize this feature.

nZambi-, 2025-02-01:
This feature is listed in the release notes for V1.10.01.51 However after updating from 1.10.01.50 it is still not available to me. 

[image]

wjyLearn, 2025-02-05:
You need to have **Developer mode turned on in preferences** to use this feature.

nZambi-, 2025-02-07:
Ugh! 

> You need to have **Developer mode turned on in preferences** to use this feature.

You are correct, Thank you. Needlessly obscure, but I appreciate it being available!  So this is addressed in V1.10.01.51. Closing.

closed 2025-02-07 by nZambi-