QIDITECH/QIDIStudio #55: Segmentation fault after sending to printer

tataranovich, 2025-01-22:
I'm using QIDIStudio_1.10.01.51_Linux_Ubuntu_20.04.AppImage on Debian 12 (Bookworm) and getting segmentation fault when sending task to QIDI Q1 Pro printer.

**Steps to reproduce:**
1. Start QIDIStudio in terminal emulator (./QIDIStudio_1.10.01.51_Linux_Ubuntu_20.04.AppImage)
1. Import model
1. Go to "Preview" tab
1. Click "Slice plate" button
1. Click "Send"
1. In "Send to Printer MicroSD card" dialog deselect "Switch to Device tab" and click "Send" button

**Result:**
Application unexpectedly terminates and the last lines in terminal are:
```
zsh: segmentation fault  ./.local/QIDIStudio_1.10.01.51_Linux_Ubuntu_20.04.AppImage
andrey:~$ ERROR: WebKit encountered an internal error. This is a WebKit bug.
./Source/WebKit/WebProcess/Network/WebLoaderStrategy.cpp(562) : internallyFailedLoadTimerFired
```

QIDITECH, 2025-01-23:
I'm sorry, but different linux versions and systems have different issues, and we can't fix them all yet.

tataranovich, 2025-02-03:
I tried the same scenario using `QIDIStudio_1.10.01.51_Linux_Ubuntu_24.04.AppImage` from Ubuntu 24.04 desktop:

```
$ grep ^VERSION= /etc/os-release
VERSION="24.04.1 LTS (Noble Numbat)"

$ ./Downloads/QIDIStudio_1.10.01.51_Linux_Ubuntu_24.04.AppImage
[skipped]
(qidi-studio:9954): Gtk-CRITICAL **: 09:07:28.302: gtk_widget_set_size_request: assertion 'width >= -1' failed
Segmentation fault (core dumped)
```

elliotnash, 2025-02-06:
This also occurs for me on both the 20.04 and 24.04 versions on Fedora 41.

wjyLearn, 2025-02-11:
In advanced versions of Linux systems, problems with WebKit cause software crashes, and a solution is being sought. For users of advanced Linux versions, you can manually add port **7125** to the device when adding the device. In this way, GCode can be sent to the device successfully without crashing the software. The Fluidd page can only be accessed through a browser first.

snigel, 2025-02-11:
I'm getting similar behavior on Arch using QIDIStudio_1.10.01.51_Linux_Ubuntu_24.04.AppImage 
Send to printer via lan -> crash

@wjyLearn I wanted to try to add port 7125, but:
Go to device tab, edit printer settings -> crash

As a workaround I save the gcode to disk and upload it via fluidd.

wjyLearn, 2025-02-11:
@snigel  You can change this by adding **":7125"** to the IP of "print host" in the configuration file, or deleting the configuration file and adding the model again in the software, adding **"7125"** to the IP when adding.

Linux physical device configuration file path **"/ home/username/config/QIDIStudio/user/default/physical_printer"**

snigel, 2025-02-28:
@wjyLearn I found the configuration file and edited it from `192.168.1.26` to `192.168.1.26:7125`.
I didn't get it to work.

- Sending the file from Qidi studio shows a progress bar that finishes, but then nothing happens on the printer.
- The device tab in Qidi studio shows "The URL can't be shown".
- Using the same ip:port in the browser displays the moonraker interface, so it's not a connectivity issue.

To anyone else editing, the path on my system is ".configuration" with a dot, ie:
`~/.config/QIDIStudio/user/default/physical_printer/`

wjyLearn, 2025-03-01:
Using port 7125 ensures that the software can successfully send GCode files to the printer without crashing. To access moonraker in a browser, use 10088 or 80, or you can access it without entering a port.

snigel, 2025-03-01:
> Using port 7125 ensures that the software can successfully send GCode files to the printer without crashing. To access moonraker in a browser, use 10088 or 80, or you can access it without entering a port.
What I'm trying to say, is that sending the file didn't work, even though it's the correct IP and port.

After adding the port, Qidi Studio shows a progress bar sending the file, but when it's done the printer doesn't print.
I tried rebooting the printer, restarting Qidi Studio and verifying IP and port, still didn't work.

dbrezack, 2025-03-03:
Segfault with `internallyFailedLoadTimerFired` 

Confirmed still exits, G-Code sends just fine, for me it's when clicking on the printer in the device menu. 

QIDIStudio_1.10.02.76_Linux_Ubuntu_20.AppImage
22.04 LTS
X plus 4 

Looks like it's happening with the webkit library. Something on the printer screen isn't playing nice. 

```
#0  0x00007ffff0f091c6 in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
#1  0x00007ffff0de419c in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
#2  0x00007ffff0f2c263 in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
#3  0x00007ffff0a56680 in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
#4  0x00007ffff0d5137b in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
#5  0x00007ffff0e658ea in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
#6  0x00007ffff0d4c445 in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
#7  0x00007ffff0d4ed63 in ?? () from /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
#8  0x00007fffefbb86ad in ?? () from /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18
#9  0x00007fffefc4ba4d in ?? () from /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18
#10 0x00007fffefc4c763 in ?? () from /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18
#11 0x00007ffff5741c44 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#12 0x00007ffff57972b8 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007ffff57412b3 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007ffff5c48d2d in gtk_main () from /lib/x86_64-linux-gnu/libgtk-3.so.0
#15 0x00005555595e1205 in wxGUIEventLoop::DoRun() ()
#16 0x00005555596f8d51 in wxEventLoopBase::Run() ()
#17 0x00005555596c7c33 in wxAppConsoleBase::OnRun() ()
#18 0x0000555559749567 in wxEntry(int&, wchar_t**) ()
#19 0x00005555569baa7b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#20 0x0000555555f87a7f in Slic3r::CLI::run(int, char**) ()
#21 0x0000555555e94c29 in main ()
```

wjyLearn, 2025-03-10:
Can temporarily refer to #8

T-McBride, 2025-11-16:
> I'm getting similar behavior on Arch using QIDIStudio_1.10.01.51_Linux_Ubuntu_24.04.AppImage Send to printer via lan -> crash
>
> [@wjyLearn](https://github.com/wjyLearn) I wanted to try to add port [...]

>  For users of advanced Linux versions, you can manually add port **7125** to the device when adding the device. 

Thank you! Been fighting this for a week, worked great.  

Related crash info if it helps: 

Linux SN-001 6.6.85-gentoo 

slic3r_main[9754]: segfault at 0 ip 00007f0cb57fb776 sp 00007fffd734c478 error 4 in libwebkit2gtk-4.1.so.0.17.10[7f0cb50bc000+38d0000] likely on CPU 0 (core 0, socket 0)                                                                                                                 
[329077.026552] Code: 00 00 00 00 48 c7 40 2c 00 00 00 00 c7 40 34 00 00 00 00 c3 f3 0f 1e fa 48 8b 47 28 48 8d 15 e1 cc ff ff 48 8b b8 68 01 00 00 <48> 8b 07 48 8b 40 20 48 39 d0 75 06 c3 0f 1f 44 00 00 ff e0 66 0f