QIDITECH/QIDIStudio #58: Camera  full view

deanra, 2025-01-24:
I updated to the new slicer studio ver. 1.10.01.51 and the camera no longer displays a full view.  only a small portion shows up on the fluidd console.

QIDITECH, 2025-01-25:
Sorry, I can't exactly imagine what you mean. Could you please provide a screenshot to explain it accurately?

deanra, 2025-01-26:
[image: page print.jpg]

QIDITECH, 2025-01-27:
Sorry, please try resending, I can't see any pictures.

deanra, 2025-01-27:
[image: page print.png]

QIDITECH, 2025-01-30:
Please do not use email forwarding, this will not display the picture.
[image]

wjyLearn, 2025-11-18:
You can turn it on or off in Settings - Camera, and then turn it on again when editing Camera Settings. Operate in this way and see if it works.