QIDITECH/QIDIStudio #60: [Feature reqest] Please allow other ports then 10088

Phil1988, 2025-01-25:
Cant establish a connection on port 80 with QidiStudio v1.10.1.51

It defaults to port 10088 when entering no port
[image]

Attempting to force port 80, it reports a working connection, but its not showing the WebUI (see in background)
[image]

Would be nice if users can use your QidiStudio and dont have to use Orca/PrusaSlicer/SuperSlicer etc.

Thanks!

QIDITECH, 2025-01-25:
Thanks for your feedback, we will optimize the window in future versions to make it easier to understand and type.

Phil1988, 2025-01-25:
Its not only about making it easier.
I want to point out, that **it is not functional**.

**You can not connect via port 80** at the tested version of QIDIStudio

edspeds, 2025-01-28:
> Thanks for your feedback, we will optimize the window in future versions to make it easier to understand and type.

Thank you this is greatly appreciated.  Will you also allow the setup of non Qidi printers?  I appreciate Qidi Studio but run a couple of different printers and would like to be able to use a common slicer.

wjyLearn, 2025-02-05:
@Phil1988  From the image, it appears that the connection has been refused. You can confirm whether this IP can be accessed normally by using a browser. If access is not possible, please check if the firewall settings are blocking the connection request.

Phil1988, 2025-02-05:
@wjyLearn Yes printer is completely accessible via this IP.
The problem is not on my side and I dont want to waste time.
Just for your information: I do know what I am doing, as I released my own firmware for this printer.

sezeryalcin, 2025-02-10:
This crashes for me when I click on printer (printer is added with no port number)
Some other forums suggested to use another port number (7150 or something ) but I could not find that post again.
Any other port numbers open on the device?

adam-hulin, 2025-02-11:
@sezeryalcin 
Port 80 - http for fluidd
Port 22 - ssh
Port 7125 - moonraker
Port 8080: mjpeg for the webcam
Port 8990 - Qidi link UDP server
Port 10088 - http for fluid ([the documentation says this is for other slicers](https://wiki.qidi3d.com/en/software/ConnectToPrinterViaWIFI))

snigel, 2025-02-11:
@sezeryalcin 
Port 80 - http for fluidd
Port 22 - ssh
Port 7125 - [moonraker](https://moonraker.readthedocs.io/en/latest/configuration/)
Port 8080 - [mjpg_streamer](https://github.com/jacksonliam/mjpg-streamer) (webcam)
Port 8990 - Qidi link UDP server
Port 10088 - http for fluid ([the documentation says this is for other slicers](https://wiki.qidi3d.com/en/software/ConnectToPrinterViaWIFI))

00stag, 2025-02-27:
I'll join in here. Please do not bind the QIDI-Studio to a port. Keep the QIDI-Studio open so that it benefits everyone who wants to use it.

Phil1988, 2025-09-16:
Its working in v2.2.0.61 if you enter 
`http://IP`

closed 2025-09-16 by Phil1988