QIDITECH/QIDIStudio #64: QIDI Studio Crash (V 1.10.1.51) Moving Application Window from one screen to another

SMRTCUBE, 2025-02-08:
Crash Log below.

QIDI Studio was opened on primary monitor (2560x1440) and was moved to second monitor (1440x2560) and was attempted to be snapped to full screen. Application crashed. This was repeatable.

`Log Name:      Application
Source:        Application Error
Date:          2025-02-08 6:08:24 PM
Event ID:      1000
Task Category: Application Crashing Events
Level:         Error
Keywords:      
Description:
Faulting application name: qidi-studio.exe, version: 1.10.1.51, time stamp: 0x6777af74
Faulting module name: QIDIStudio.dll, version: 0.0.0.0, time stamp: 0x678dae46
Exception code: 0xc000041d
Fault offset: 0x0000000000cf6159
Faulting process id: 0x58B0
Faulting application start time: 0x1DB7A7E53D482D5
Faulting application path: C:\Program Files\QIDIStudio\qidi-studio.exe
Faulting module path: C:\Program Files\QIDIStudio\QIDIStudio.dll
Report Id: 229e2cab-fbc2-461f-b558-86bd797657f0
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" Guid="{a0e9b465-b939-57d7-b27d-95d8e925ff57}" />
    <EventID>1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x8000000000000000</Keywords>
    <TimeCreated SystemTime="2025-02-08T23:08:24.2022848Z" />
    <EventRecordID>16554</EventRecordID>
    <Correlation />
    <Execution ProcessID="5268" ThreadID="20292" />
    <Channel>Application</Channel>
  </System>
  <EventData>
    <Data Name="AppName">qidi-studio.exe</Data>
    <Data Name="AppVersion">1.10.1.51</Data>
    <Data Name="AppTimeStamp">6777af74</Data>
    <Data Name="ModuleName">QIDIStudio.dll</Data>
    <Data Name="ModuleVersion">0.0.0.0</Data>
    <Data Name="ModuleTimeStamp">678dae46</Data>
    <Data Name="ExceptionCode">c000041d</Data>
    <Data Name="FaultingOffset">0000000000cf6159</Data>
    <Data Name="ProcessId">0x58b0</Data>
    <Data Name="ProcessCreationTime">0x1db7a7e53d482d5</Data>
    <Data Name="AppPath">C:\Program Files\QIDIStudio\qidi-studio.exe</Data>
    <Data Name="ModulePath">C:\Program Files\QIDIStudio\QIDIStudio.dll</Data>
    <Data Name="IntegratorReportId">229e2cab-fbc2-461f-b558-86bd797657f0</Data>
    <Data Name="PackageFullName">
    </Data>
    <Data Name="PackageRelativeAppId">
    </Data>
  </EventData>
</Event>`

```

wjyLearn, 2025-02-10:
QIDIStudio is currently not suitable for dual-screen mobile use. Compatibility may be considered in subsequent optimizations.

snigel, 2025-02-11:
Is this a duplicate of #7 ?

W0lpertinger, 2025-02-20:
Same here, but do you need dual-screen support to just move the Window to an other screen?
And where is it gone?
It's based on Bambu Studio and Prusa Slicer, both and Orca Slicer have no problem moving to an other monitor.

closed 2026-11-05 by wjyLearn