QIDITECH/QIDIStudio #68: Project saved incorrectly when Post processing script field contains quotes (")

Rudokhvist, 2025-02-21:
# Environment

OS: Windows 10
QIDIStudio Version: 1.10.1.51
Printer: QIDI Plus4

# Steps to reproduce:

1. Create new project
2. Right click on plate - > add part - > Cylinder
3. Right click on cylinder - > Fill bad with copies
4. Add text with quotes to Post Processing Script field (Other tab). Example: `"C:\test script.bat"`
5. Save project
6. Open project you just saved

# Expected result:

Project opens correctly and everything looks same

# Actual result:
Error box displayed:

[image]
Project opens, but place size is incorrect, and part of objects is outside of plate

# Additional notes

Reproduction steps 1-3 only needed for demonstration purposes, actual issue is caused by step 4.
Quotes in Post Processing Script field are needed for ability to use paths with spaces, without quotes it does not work for such paths, but with quotes everything works as expected.
Possible ways to fix: 
- somehow escape text from Post Processing Script field before saving project, so that file was saved correctly
- check the Post Processing Script field and show error when attempting to save if it contains quotes (it's not really nice, but it's not a big deal. But the problem is that it might happen from other characters too, so escaping is more reliable).

# Thank you for your time!

wjyLearn, 2025-02-22:
Thanks for the feedback, could you provide your test script so that we can reproduce the problem?

Rudokhvist, 2025-02-22:
Just FYI:
Neither OrcaSlicer nor PrusaSlicer have this issue. But if I save 3mf from them and open it in QIDIStudio I'm still getting invalid plate size. If I save project created with reproduction steps above from QIDIStudio - it opens correctly in OrcaSlicer, and also opens in PrusaSlicer _almost_ correctly, but the post-processing script is not loaded for some reason (that's not your problem obviously). So from my observations it seems that this issue happens upon loading, not upon saving. 
I also tried Slic3r as a common source of all the above, but it does not seem to save post-processing scripts in 3mf files, and thus can't have this issue (also, it does not work correctly with quoted path, but allows escaping manually with exclamation mark `!` for spaces, etc).

Rudokhvist, 2025-02-24:
> I have found a solution to a similar problem to yours, and I wonder if I can solve your problem. Refer to [prusa3d/PrusaSlicer#3335](https://github.com/prusa3d/PrusaSlicer/issues/3335)

In the linked issue user had problems with their post processing script, not with slicer itself.

wjyLearn, 2025-02-24:
I roughly understand what you mean: If the post-processing script path contains spaces (for example, C:\Users\17826\Desktop\tests cript.bat;), an error will be prompted during slicing. If the script path is enclosed in quotation marks (for example, "C:\Users\17826\Desktop\tests cript.bat";), the script can execute normally during slicing. However, after saving and then reading the model file again, the post-processing script configuration included in the model cannot be read.
You can temporarily use the path expression without quotation marks (do not include spaces in the path). We will make the path expression you proposed compatible in future versions.

Rudokhvist, 2025-02-24:
> I roughly understand what you mean: If the post-processing script path contains spaces (for example, C:\Users\17826\Desktop\tests cript.bat;), an error will be prompted during slicing. If the script [...]

Yes, that's exactly how it is, and that's exactly how I circumvent this issue for now, just decided to report it since it's clearly a bug. Thank you for taking it into consideration!

Rudokhvist, 2025-05-13:
Still not fixed in V2.00.03.54

Rudokhvist, 2025-08-07:
Update for V2.2.0.60:

Still not fixed, but behavior changed: no error is displayed, instead QIDI studio only loads geometry data from .3mf file, without any settings. For comparison, OrcaSlicer opens same file correctly. Please fix.

closed 2025-08-19 by wjyLearn