QIDITECH/QIDIStudio #70: Add Custom G-code is disabled

compf99, 2025-02-27:
Hi, why "Add Custom G-code" is disabled.  I've searched everywhere and can't find anywhere where I can activate it.

You are doing a very good job, thank you for what you are doing, thank you also for the previous answers!

Qidi Q1 PRO

wjyLearn, 2025-02-28:
Not sure which version of the software you are using? In the QIDIStudioV1.10.01.50 version, there is a "Add custom G-code" button.

[image]

compf99, 2025-02-28:
Thanks, you don't have to apologize. Is there a setting in a file or windows registry that disabled the 2 options?

wjyLearn, 2025-02-28:
The setting for disabling 2 options should not be available. Can you describe the configuration of your computer and provide a screenshot of the "no button" situation?

compf99, 2025-02-28:
Windows 11 PRO, 64GB ram, i9 proc, SSD M2

[image]

wjyLearn, 2025-02-28:
Both of these features are not available with object-by-object printing. You can modify the printing order to print layer by layer.

[image]

compf99, 2025-02-28:
"By layer" activated the 2 options but it doesn't help me in what I want to do. Specifically I want to insert custom Gcode before printing each object. And when it will be the case for each object where it is still necessary. It's a senseless constraint only when in "By layer" mode. Maybe in the next update you will remove this constraint. Thank you for all !