QIDITECH/QIDIStudio #71: Doesnt start "Loading Configuration..." Windows 10 V 1.10.1.51

dalexand8, 2025-02-27:
QIDI Studio is just shows "Loading Configuration..." then goes away. Nothing happens. I have uninstalled multiple times. 

Windows 10 22H2

QIDI Studio - V 1.10.1.51

[image]

wjyLearn, 2025-02-28:
There may be a configuration error, you can package and upload the configuration, which will help us to help you resolve the problem. You can then delete the configuration and try to open QIDIStudio. Configuration file address: **C:\Users\username\AppData\Roaming**.

dalexand8, 2025-02-28:
Thank you that got it going again! 

[QIDIStudio-NotWorking.zip](https://github.com/user-attachments/files/19018404/QIDIStudio-NotWorking.zip)