QIDITECH/QIDIStudio #72: QIDIStudio Crashes M1 Mac after Launching.  Spiral Mode only works when wall loop is 1.

DanielJMorel, 2025-03-01:
The last time that I successfully had QidiStudio open, I had performed a Volumetric Velocity Calibration.After that, Launching QidiStudio crashes when launched.    I have uninstalled QidiStudio, rebooted, and re-installed version 01.10.01.51 but it still crashes.  I get a dialogue box that says "Spiral Mode only works when wall loop is 1..." Whether I select Yes or No, QidiStudio crashes and brings up the following report:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               QIDIStudio [1039]
Path:                  /Applications/QIDIStudio.app/Contents/MacOS/QIDIStudio
Identifier:            com.qiditech.qidi-studio
Version:               01.10.01.51 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-03-01 13:18:23.7697 -0500
OS Version:            macOS 13.5.2 (22G91)
Report Version:        12
Anonymous UUID:        5A24B9A9-17C9-23CD-7AFE-55AF5E3D92DA

Time Awake Since Boot: 2100 seconds

System Integrity Protection: enabled

Crashed Thread:        0  qidistu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000058
Exception Codes:       0x0000000000000001, 0x0000000000000058

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [1039]

VM Region Info: 0x58 is not in any region.  Bytes before following region: 105553518919592
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      MALLOC_NANO (reserved)   600018000000-600020000000 [128.0M] rw-/rwx SM=NUL  ...(unallocated)

Kernel Triage:
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage

Thread 0 Crashed:: qidistu_main Dispatch queue: com.apple.main-thread
0   QIDIStudio                    	       0x1056acedc Slic3r::GUI::Tab::on_value_change(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, boost::any const&) + 11848
1   QIDIStudio                    	       0x1057f7b3c Slic3r::GUI::ConfigManipulation::update_print_fff_config(Slic3r::DynamicPrintConfig*, bool, bool) + 9080
2   QIDIStudio                    	       0x1056bc8d8 Slic3r::GUI::TabPrint::update() + 648
3   QIDIStudio                    	       0x1056aed68 Slic3r::GUI::Tab::load_current_preset() + 548
4   QIDIStudio                    	       0x1054ac8d4 Slic3r::GUI::GUI_App::load_current_presets(bool, bool) + 296
5   QIDIStudio                    	       0x1054a73a4 Slic3r::GUI::GUI_App::on_init_inner() + 5936
6   QIDIStudio                    	       0x1054a5b60 Slic3r::GUI::GUI_App::OnInit() + 24
7   QIDIStudio                    	       0x1076947d0 wxApp::CallOnInit() + 152
8   QIDIStudio                    	       0x107782818 wxEntry(int&, wchar_t**) + 144
9   QIDIStudio                    	       0x1054908f8 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 512
10  QIDIStudio                    	       0x10469ca38 Slic3r::CLI::run(int, char**) + 11936
11  QIDIStudio                    	       0x1046febe4 main + 172
12  dyld                          	       0x18a8a7f28 start + 2236

Thread 1::  Dispatch queue: com.apple.root.user-interactive-qos
0   libsystem_kernel.dylib        	       0x18abbff54 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x18abd2280 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x18abc8bb8 mach_msg_overwrite + 604
3   libsystem_kernel.dylib        	       0x18abc02d0 mach_msg + 24
4   CoreFoundation                	       0x18acde7e4 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x18acdd0c4 __CFRunLoopRun + 1208
6   CoreFoundation                	       0x18acdc4b8 CFRunLoopRunSpecific + 612
7   Foundation                    	       0x18bc55fbc -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 212
8   AppKit                        	       0x18e279a98 -[NSAnimation _runBlocking] + 312
[... 445 lines cut ...]
        "MYRIAD_BOOSTS" : "62fe74805312cd4599bd3c81"
      },
      "deploymentId" : 240000006
    }
  ],
  "experiments" : [

  ]
}
}

Model: Macmini9,1, BootROM 8422.141.2, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: Roku TV, 3840 x 2160 (2160p/4K UHD 1 - Ultra High Definition), Main, MirrorOff, Online
Memory Module: LPDDR4, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: May 13 2023 07:20:48 version 18.20.383.15.7.8.150 FWID 01-b37727a5
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
USB Device: USB30Bus
USB Device: ANT USB-m Stick
USB Device: ANT USB-m Stick
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.

wjyLearn, 2025-03-03:
There should be a problem with your configuration file, you can package and upload the configuration file so that we can reproduce the problem. Then delete all the old configuration files and start QIDIStudio to use.

DanielJMorel, 2025-03-03:
My apologies I'm 2 weeks into QiDiStudio.  Can you instruct me on how to get the configuration files?

wjyLearn, 2025-03-04:
The address of the configuration file is **C:\Users\Username\AppData\Roaming\QIDIStudio**. AppData requires a visualization of hidden folders to be turned on in order to access them.

DanielJMorel, 2025-03-07:
> wjyLearn

Sorry, I am on an apple computer.  I  don't see that directory even when I look under my users directory.  Can you tell me the path for an apple computer?

wjyLearn, 2025-03-07:
Oh, I'm sorry. I didn't read the error message carefully. Apple computer's path is a **Users/Username/Library/Application Support/QIDIStudio**. Remember to package and upload the old configuration before you delete it so that we can reproduce your problem.

DanielJMorel, 2025-03-07:
Found it.  Thank you.  When you say, “Package and upload”, do you mean to compress the QidiStudio folder.  do you want me to attach it to the email or is there another method you would like me to use?  It is a pretty big compressed file.  Actually it is only 8MB zipped up.

DanielJMorel, 2025-03-07:
Deleting the Users/Username/Library/Application Support/QIDIStudio folder worked.  I am now able to use QidiStudio.

Thank you and let me know if you should need anything else.

closed 2025-03-07 by DanielJMorel