QIDITECH/QIDIStudio #74: QIDITech github lacks repositories required to build QIDIStudio.

Rudokhvist, 2025-03-04:
You guys sure bothered to auto-replace bambu credentials to yours, and it's understandable, but now the source code don't build, because it lacks those repositories:

https://github.com/qiditech/ffmpeg_prebuilts/releases/download/7.0.2/7.0.2_msvc.zip
https://github.com/qiditech/gmp/archive/refs/tags/6.2.1.tar.gz
https://github.com/qiditech/wxWidgets
https://github.com/qiditech/boost/releases/download/1.78.0/boost_1_78_0.zip

Either revert those changes or (better) create forks of those repositories under QIDITECH user. Right now you are violating rules of GNU license this code is licensed under, because by license it's REQUIRED to disclose source of your binaries, and the source code in this repository can't be build, so it clearly NOT the code your binaries use. This may cause serious legal issues for you if you won't address it (not from me of course, I'm your customer and I'm interested in your prosperity, but I'm sure bambu will be happy to cause you some troubles).

QIDITECH, 2025-03-06:
Thanks for your feedback, we will fix it in the next version.

Xephier102, 2025-04-19:
> Thanks for your feedback, we will fix it in the next version.

And how long will that be? You replied to this over a month ago, and the last update to the repo was two months ago, and I bought my qidi Q1 pro 3 days ago, and haven't even used it yet due to the headaches in just getting a software that works, and will connect to this printer..

wjyLearn, 2025-04-21:
The corresponding update will be implemented in the next version. As for the specific update date, it has not been determined yet.

Xephier102, 2025-04-21:
> The corresponding update will be implemented in the next version. As for the specific update date, it has not been determined yet.

Thankfully I'm smart enough to find workarounds, but I'm betting that a lot of people are just getting frustrated and returning their printers. 

Turns out that if you just go to your browser and type http://10.12.0.94 or whatever the IP is of your printer, it opens up an interface to your printer which allows you to upload models directly to it via the 'jobs' tab on the side. 

From there, you can just use any slicer software(with qidi presets) to slice the model and save as gcode, then upload that with the above method. Not as simple as doing it with one software, but it gets the job done.