QIDITECH/QIDIStudio #76: Login button non-functioning in QIDIStudio v1.10.2.76

TimFeat, 2025-03-07:
The login button appears to be non-functional on the password login tab.
Same issue if I use the Forgot Password button,

wjyLearn, 2025-03-08:
You mean when you click the login button on the login page, there is no response? You can try to remove the cache and then try to use the login function again. Specific how to clear the cache can refer to [wiki] (https://wiki.qidi3d.com/en/software/qidi-studio/troubleshooting/blank-page)

drathbone, 2025-03-21:
Just to add I'm having the same issue - the login button (either on the Password Login or Verify Login pages) does nothing - it's not a time delay as I've left it for >10mins. 

Despite this being on a clean download/install of the software (first time Qidi user and only got the printer a day or two ago), I've also tried the clear cache process, which makes no difference.

wjyLearn, 2025-03-22:
> Just to add I'm having the same issue - the login button (either on the Password Login or Verify Login pages) does nothing - it's not a time delay as I've left it for >10mins.
>
> Despite this being on [...]

This phenomenon of yours is caused by poor network. You can reopen the login page and perform the login operation.

lishniy, 2025-04-17:
I have this problem too. Click on login does nothing.

Xephier102, 2025-04-19:
> Perhaps sometimes, due to network or proxy issues, clicking "login" does not perform any operation. If the network proxy is on, you can try to turn it off before performing the login operation. If the proxy is not enabled, it might be caused by network fluctuations. You can try opening the login page several times and performing the login operation.

I can confirm that this is not a network issue. Though potentially a wrong password issue. Cuz one thing that's definitely wrong in the code, is that if you enter a wrong password, it does nothing at all. It acts as if you never hit the button, no refresh of the window/screen, no 'wrong password' dialog, nothing at all. 

While long-time Linux users may shrug this off, cuz they're used to entering commands into the terminal all day and being given zero feedback. But, long-time windows/mac users will have more to say.  

I'm personally dealing with a printer connection issue now though. Manage to connect my printer, and even shows the camera feed for my printer for a second, but freezes immediately and crashes seconds later, and the frozen window makes some wrong statement about klipper not being on, and a button there to 'restart klipper' but can't hit it since the software is already frozen by the time you see it. 

The app on my phone connects just fine, but the one on my computer, not yet.. It's just an appimage, so it's not like I messed up installing it, though I am using Linux Mint, and not Ubuntu, so that could be a thing. anywho, gonna add this to it's own post, just added it here while responding to you to see if I get any sooner response, since the response times in  this issue section seem to really drag out.

wjyLearn, 2025-04-21:
Regarding your issue of freezing the window, you can temporarily refer to the solution in #8.

LeviMThomas, 2025-09-07:
Same. On my M4 Macbook Pro up to date with Qidi Studio up to date (sep, 7, 2025) When I go into printers min is there just fine its loggin the app into my profile by means of password or link. Crazier yet, the send code to email works perfectly but when you type it in the Login button just doesn't click.