QIDITECH/QIDIStudio #77: Dragging STLs into window no longer works after last update

arketec, 2025-03-08:
Prior versions allowed users to import STL and other files by dragging from Windows Explorer into the project window. This no longer works and one must use File > Import to bring in STLs.

gwideman, 2025-03-09:
Oddly, I was used to STL drag-drop working in Studio 1.10.1.50. Then I installed 1.10.2.76 which forgot about my Plus4 Device entry, and was missing a way to configure a Plus4, so I had to uninstalled it.  When I reinstalled 1.10.1.50, the STL drag-drop functionality no longer worked. 

This sequence of events leading to the drag-drop functionality disappearing from a version in which it worked may be an additional clue as to the root cause.

arketec, 2025-03-09:
Strangely, it started working again after several restarts. A single restart did not solve it, but i have restarted it a few times since and suddenly it is allowing drag and drops again. Not sure what is going on, but that is rather strange behavior

wjyLearn, 2025-03-10:
Perhaps you need To check the** Assocaite Files To QIDI Studio** option in **Preferences**.

[image]

arketec, 2025-03-10:
Thanks, but that would be to solve a different problem. That is the solution to double-clicking an STL or other file and have it open in Qidi Studio. I was referring to drag-and-droping STL files into Qidi Studio. Also, the problem seems to have fixed itself, so I am not sure what changed as it drag-and-drop does now work again for me. 

I am only keeping this open as there still might be a bug since another person experienced the same problem. I can close it if that is what is desired

wjyLearn, 2025-03-10:
Thank you for your feedback, we will pay attention to this issue in the future testing process.

kamermans, 2025-05-31:
Just dropping a note to say that this is one of my greatest annoyances with the latest updates (I noticed it when switching to 2.x).  I think it's an important feature because a typical workflow, at least for me, is to export an STL from my CAD program, which I then see in Explorer/Finder (I'm on Windows), and then I drag it and drop it into QIDI Studio.  Without this functionality I'm forced to use the file browser to find my model, and that means sifting through hundreds of folders, many levels deep.

wjyLearn, 2025-06-02:
Thank you for your feedback. We will do our best to find out the cause of the problem and fix it in the subsequent version.