QIDITECH/QIDIStudio #78: Qidi Studio 1.10.2.76 is not equipped to work with Plus4?

gwideman, 2025-03-09:
I previously had Studio 1.10.1.50 working very well with my Qidi Plus4.   Then prompted by Studio's "new release" message I downloaded and installed 1.10.2.76.  
Problems:

- When that new version started up, the device list was empty, specifically it had forgotten the entry for my existing Plus4 printer.
- When I attempted to create a new entry for my printer, the "Choice your physical printer preset" list lacks any preset for Plus 4. So I could not add an item for my printer.

I then completely uninstalled Qidi Studio, and reinstalled version 1.10.1.50.

- That old version DID have a Devices entry for my Plus4 printer (no need to create a new Device entry)

Conclusion:
Some data or code relating to Qidi Plus4 is missing from 1.10.2.76.

wjyLearn, 2025-03-10:
The preset for Plus4 is missing, you can add it by opening the configuration wizard.

As for the problem you mentioned, we have not encountered it in the test. It would be great if you could upload your configuration file to help us reproduce the problem.

gwideman, 2025-03-10:
> The preset for Plus4 is missing, you can add it by opening the configuration wizard.
> 
> As for the problem you mentioned, we have not encountered it in the test. It would be great if you could upload your configuration file to help us reproduce the problem.

After reverting to 1.10.1.50 and using it, I tried installing 1.10.2.76 again.

I could find no way to open a "Configuration Wizard".  However, I eventually found the gear icon next to Printer heading on the Preview page, and that led to a dialog that allowed selecting printer types.  There I selected "X-Plus 4".   

Having done that, the Device page still did not load or show the existing printer settings from 1.10.1.50.  However, the Device > Local > "+" (add) button dialog does now show X-Plus 4 as an option.

At that point I was able to add my Plus 4, and it appears to communicate OK, show the camera and so on.

But why is this necessary?  The new version of Studio should just be able to use the existing settings.

wjyLearn, 2025-03-11:
The new version usually inherits the old version of the configuration file, I am not sure why this issue is happening to you. If you encounter a similar problem next time, you can upload the configuration file, which will help us solve the problem better.

You can open a Configuration file by going to **The dropdown button next to the file button-help-show Configuration folder**.