QIDITECH/QIDIStudio #79: Filename format setting not working anymore in 1.10.2.76

dssamusaran, 2025-03-10:
The custom filename format is no more applying either when sending the gcode to a machine, or printing the gcode directly.

It does work when exporting the gcode.

In prior version it was working for the send/print options without issues.

This is a fundamental feature, this revision is not usable with this state as it implies a lot of risks when prototyping with a lot of different but very similar gcodes. Simply showing the stl name won't cut it.

Issue was reproduced on three different computers. It is linked to the new version.

Example when Sending:

[image]

Example when Printing:

[image]

Example when exporting gcode, with the expected name working fine:

[image]

wjyLearn, 2025-03-11:
Thank you for your feedback, we will fix this issue in the next edition.

dssamusaran, 2025-03-11:
Thanks!

And for the rest, thanks for your continuous support and updates. Great work!

DU4FDR, 2025-03-26:
Bonjour,
merci de corriger le nom de fichier pour implémenter un dossier et de gérer les documents par dossier.
A+