QIDITECH/QIDIStudio #80: Can not select Plate Type after installing 1.10.2.76

benzok, 2025-03-14:
Yesterday I installed the new version and after that the "Plate Type" selection is grayed out and if  I open a filament the only Bed Temperature available is for the "Textured PEI Plate"

Attached is the log-file from the first start of  1.10.2.76

Please advise,

[image]

[image]

[debug_Thu_Mar_13_16_48_36_6559.log](https://github.com/user-attachments/files/19243932/debug_Thu_Mar_13_16_48_36_6559.log)

wjyLearn, 2025-03-15:
You will need to turn on printer Settings - check **Support multi bed types**.

benzok, 2025-03-15:
That was it !  Thank you for your prompt reply

closed 2025-03-15 by benzok