QIDITECH/QIDIStudio #81: Vector too long

Sundlale, 2025-03-16:
Hi I have created a few stl files from Blender.

If I import them to QidiStudio and slice them they are ok. 

If I turn the model upside does (exactly 180 degrees so it sits properly on the bed) I get an error "Vector Too Long".

Does anyone else get this and/or know the reason solution?

Printer is Plus4.
Blender 4.3.2.

TIA

wjyLearn, 2025-03-17:
Can you provide a model for reporting errors? So that we can find the cause of the problem.

Sundlale, 2025-03-17:
I tried to attach the stl but it says "file Type not allowed:.

I have changed the extension to .txt so you can change it back to .stl and it should work.

[seive 26 - Copy.txt](https://github.com/user-attachments/files/19276914/seive.26.-.Copy.txt)

Sundlale, 2025-03-17:
I have also just received this error when trying to rotate the stl.

[image]

Sundlale, 2025-03-17:
It seems to slice fine in Cura.

[image]

wjyLearn, 2025-03-17:
Using the model you provided, we tested it. There was no problem as you described. I don't know whether our operation is different?

[image]

Sundlale, 2025-03-17:
Thank you for taking the trouble. Is there a way of sending me a gcode file
that i can send directly to the printer.

wjyLearn, 2025-03-18:
[seive.26.-.Copy.txt](https://github.com/user-attachments/files/19303036/seive.26.-.Copy.txt)

wjyLearn, 2025-03-18:
[seive.26.-.Copy.txt](https://github.com/user-attachments/files/19303079/seive.26.-.Copy.txt)

Sundlale, 2025-03-19:
I need your help on this please.
I downloaded the file seive.26.-.Copy.txt and that appears to be an STL file.
I converted the extension to STL and tried to load that in to QidiStudio and got this 

[image]

[image]

so I'm probably doing something wrong.

Are you able to save it as a .3mf file so I can import it directly?
I am using a Plus 4 but I don't need to print directly.
If I can open the project I can at least have a look

Sundlale, 2025-03-19:
It does not happen every day but I also get this message with a lot of files I load,

[image]
Any things I should look at to avoid this please?

wjyLearn, 2025-03-19:
I uploaded a.gcode file before. So if you change it to.stl file, the information will not be read. The file uploaded this time is in.3mf format.

[seive.26.-.Copy.3mf.txt](https://github.com/user-attachments/files/19339542/seive.26.-.Copy.3mf.txt)

wjyLearn, 2025-03-19:
> 它不是每天都会发生，但我也会收到这条消息，其中包含我加载的很多文件，
> 
> [image]请问我应该看什么东西来避免这种情况？

Perhaps you can tell us the configuration of your computer and upload the configuration file so that we can try to reproduce the problem.

wjyLearn, 2025-03-20:
> It does not happen every day but I also get this message with a lot of files I load,
> 
> [image] Any things I should look at to avoid this please?

Perhaps you can tell us the configuration of your computer and upload the configuration file so that we can try to reproduce the problem.