QIDITECH/QIDIStudio #84: Moving Qidi studio to second screen makes the application crash

JoopVerdoorn, 2025-04-19:
- Qidi studio version V1.10.02.76
 
- Windows version:
Language: Dutch
Editie	Windows 11 Pro
Versie	24H2
Installatiedatum	‎11-‎1-‎2025
Build van besturingssysteem	26100.3775
Ervaring	Windows Feature Experience Pack 1000.26100.66.0

- Hardware PC:
Processor	AMD Ryzen 7 5700X 8-Core Processor                3.40 GHz
Geïnstalleerd RAM-geheugen	32,0 GB
Graphics     NVIDIA GeForce GTX 1050
Main monitor: 4k
2nd monitor: UHD

- Printer
Q1 pro

- Expected behaviour
When moving the Qidi studio screen from the main monitor to the 2nd monitor it works

- Expercienced behaviour:
Qidi studio freezes and crashes after appr. a minute. This doesn't happen with Orca slicer

wjyLearn, 2025-04-21:
It has been confirmed that if the graphics card or resolution or PDI is different between the two screens, the software may not work properly on the second screen. But no solution has been found, please wait.

snigel, 2025-05-10:
There are two similar tickets; #7 and #64