QIDITECH/QIDIStudio #90: The configured Pressure Advance in filament settings are not being aplied in the printer when starting the print

sneik15, 2025-05-26:
The configured Pressure Advance in filament settings are not being aplied in the printer when starting the print, for example:

[image]

and in the printer:

[image]

Printer: Qidi Plus 4
SW Version: v2.00.03.55

sneik15, 2025-05-27:
I know that, but right now that's not happening, if I put 0.05 in filament settings it's not being applied and i can see that in the printed parts, if for example i do the same thing in Orca slicer the pressure advance update with no problem

sneik15, 2025-05-27:
GCODE in Qidi Studio:
[image]

GCODE in Orca Slicer:
[image]

wjyLearn, 2025-05-28:
Thank you very much for your feedback. We will fix this issue in the subsequent version.

sneik15, 2025-05-30:
Thx, Fixed in v2.00.03.56