QIDITECH/QIDIStudio #94: Potential Sizing / Rotation Bug

nicholasbattles, 2025-06-11:
Cloned rotated object dimension discrepancy. The Part Name menu dimensions do not align to the part coordinates dimensions when the part is rotated. The part coordinate dimensions represent the correct dimensions, the part name menu dimensions are incorrect. Image is provided for reference. 

[image]

Notice when the cloned part is not rotated, the size dimensions are accurately represented in the part name menu. 

[image]

nicholasbattles, 2025-06-12:
Why would the X coordinate dimension stay the same but Y and Z dimensions change?

wjyLearn, 2025-06-13:
Observed from the perspective of world coordinates, after rotating the model, the size of the model in three-dimensional coordinates will change accordingly. For example, a cuboid with a length, width and height of 10, 3 and 5 respectively. The corresponding world coordinates x is 10, y is 3, and z is 5. When flipped 90 degrees along the X-axis, z will become 3 and y will become 5.

The object coordinates are established in a coordinate system based on the model itself and are related to the model size.

In fact, inversion and displacement do not change the true size of the model.