QIDITECH/QIDIStudio #97: Software crash after freeze on monitor drag (v2.00.03.56)

Simone3D, 2025-06-22:
Description:
The Qidi slicing software (version v2.00.03.56) consistently freezes for several seconds, then crashes, immediately after its window is dragged from the primary display to a secondary display.

Steps to Reproduce:
1.  Open Qidi Slicing Software v2.00.03.56 on the primary monitor.
2. Click and drag the software window towards the secondary monitor.
3.  The software freezes and becomes unresponsive for a few seconds.
4. After this brief freeze, the software crashes completely.

System Information:
Operating System: Windows 10
Primary Monitor Resolution: 1920x1080
Secondary Monitor Resolution: 1920x1080

Thanks

dsieradzki, 2025-07-31:
Unfortunately this doesn't work either, after setting compatibility mode app look blurry and controls are not clickable. 
Simply add possibility to resize this panel like OrcaSlicer (I know that QIDIStudio is based on Bambu).

wjyLearn, 2025-08-01:
We will optimize the sidebar scaling feature in the next version.