QIDITECH/QIDIStudio #102: Prime Tower Miscalculation

Meg1Print, 2025-07-24:
while trying to print multi color parts with a color change in the same layer there is always an error while printing

[image]

so - no klipper printer on this earth can parse a g-code file with digits after a comma - it is not supported. this error only occurs, when the prime tower is activated and is generated by the max. speed settings

[image]

to generate the g-code files before printing and delete the comma and the digits by hand may be a workaround for small prints, but with larger prints it is a nightmare.
It is a great disadvantage when you'r in multi color printing....

wjyLearn, 2025-08-11:
Thank you for the feedback. After our testing, we confirmed that the bug is related to the German system. We will fix it in the upcoming version.

wjyLearn, 2025-08-19:
It has been fixed in the latest version.

closed 2025-08-19 by wjyLearn