QIDITECH/QIDIStudio #108: Sync with box not working over Wi-Fi

StellarSolstice, 2025-08-07:
Printer version 1.7.1
Studio version 2.02.00.60

[image]

Only issue over Wi-Fi, sync with box over Link works

[image]

Printer is connected over Wi-Fi in device tab and box is recognized in fluidd

wjyLearn, 2025-08-08:
You can restart the machine and then try connecting to the box from the software again.

docangle, 2025-08-08:
I am having the same issue with LAN, WiFi, and Link.  Always says, "The printer has not connect to box, please check."  I am wondering if it is a Port issue since I am using Unbiquity.  I also have a user account set up in fluidd with api key, but this should not be an issue, I can access the fluidd page from the device tab as well as send prints without issue.

thx1200, 2025-08-11:
Having this issue too, removed authentication requirement and it works.  Looking forward to the patch.