QIDITECH/QIDIStudio #111: [BUG] Setting Box temperature in filament profile doesn't work

dsieradzki, 2025-08-07:
When I try to set the Box Temperature in the filament profile, it doesn't work.

I've checked the G-code generated by QIDI Studio (for PETG), and it includes the command:
BOX_TEMP_SET VT0=44,
but the printer doesn't activate the box heater.

On the other hand, when I manually enable it using:
SET_HEATER_TEMPERATURE HEATER=heater_box1 TARGET=44,
it works as expected — but that's not the command the slicer generates.

In my opinion, the Box Temperature tooltip should be more informative for new users. It should mention the "QIDI Box multicolor system." At first glance, I was confused — I wondered "what box?" and "how is it different from chamber temperature?"

wjyLearn, 2025-08-08:
Since different filaments require different temperatures in a multi-color box, if a multi-color box contains multiple types of filaments (such as PETG and PLA), the temperature setting for the multi-color box will use the lowest value between the two.  And SET_HEATER_TEMPERATURE HEATER=heater_box1 TARGET=44. This command will force a startup.

Additionally, your suggestion is good, and we will optimize the prompt messages for the multi-color box system in the next version.

dsieradzki, 2025-08-08:
"Since different filaments require different temperatures in a multi-color box, if a multi-color box contains multiple types of filaments (such as PETG and PLA), the temperature setting for the multi-color box will use the lowest value between the two"

This information should also be added to the field description fox "Box temperature".

If it works this way, the box heater will not be enabled when PLA or TPU is inside the box, due to the fact that in the official_filas_list.cfg, the values for PLA and TPU are set to zero.

Please update the filament temperatures also.

wjyLearn, 2025-08-08:
Thank you for your feedback. The corresponding information will be updated in the next version.